stephenlacy/msi-keyboard-gui #2: Don't  work on arch

Turbito, 2016-06-03:
I try to execute it, the window opens but dont show options

My arch is updated

RicardoEPRodrigues, 2016-06-03:
Same here.

Ubuntu 16.04
npm 3.9.5

Betree, 2016-06-03:
Same here with a GE72 6QD Apache Pro on Mint 17.3. NPM 3.8.9

[image]

CompMedChem, 2016-06-10:
Same is happening to me
MSI GS60
Ubuntu 16.04
npm 3.5.2

MarcusLlewellyn, 2016-06-12:
First, this is awesome! Thank you! Same issue as the others though.
`MSI GE72 6QF`
`Ubuntu 16.04 LTS`
`Kernel 4.4.0-24-generic`
`node v4.2.6`
`npm 3.5.2`
`libusb-dev 0.1.12`

stevelacy, 2016-06-20:
Please open the dev tools (Ctrl + Shift + i ) and see if there are any errors.

stevelacy, 2016-06-20:
Actually - build from source (Development) and see if it runs. There seems to be an error with the compiled package

MarcusLlewellyn, 2016-06-25:
Sadly, Ctrl-Shift-i has no effect on my system.

So, to the source! But I'm having trouble building it. `npm install` fails with the following error:

```
npm WARN msi-keyboard-gui@0.1.1 No repository field.
npm WARN msi-keyboard-gui@0.1.1 No license field.
npm ERR! Linux 4.4.0-24-generic
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "node-hid"
npm ERR! node v4.2.6
npm ERR! npm  v3.5.2
npm ERR! file sh
npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn

npm ERR! node-hid@0.5.1 install: `node-pre-gyp install --fallback-to-build`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the node-hid@0.5.1 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the node-hid package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs node-hid
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls node-hid
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/mikey/src/msi-keyboard-gui/npm-debug.log
```

It looks as though someone has filed a similar issue with node-hid over at: [node-hid/node-hid issue #144](https://github.com/node-hid/node-hid/issues/144)

MarcusLlewellyn, 2016-06-25:
Okay, upgrading node to v.4.4.6 via the node.js repo allowed me to build successfully. Unfortunately, I still have a blank window. Happily, Ctrl-Shift-i now works. There is one error:

```
module.js:440 Uncaught Error: Cannot find module '/home/mikey/src/msi-keyboard-gui/node_modules/msi-keyboard/node_modules/node-hid/build/HID/v0.5.1/Release/node-v48-linux-x64/HID.node'
```

stevelacy, 2016-06-25:
To fix the node-hid error this should work: `rm -rf node_modules && npm i`
Then run `npm start`
That should rebuild the modules for electron and start the app

MarcusLlewellyn, 2016-06-25:
Nope, still a blank screen with the same error. When I dig into the directory structure, there's a mismatch. Where the app is looking in `node-v48-linux-x64` for HID.node, it is instead being placed in `electron-v1.1-linux-x64`. So the complete path I have to HID.node is:
`~/src/msi-keyboard-gui/node_modules/node-hid/build/HID/v0.5.1/Release/electron-v1.1-linux-x64/HID.node`

chrisranderson, 2016-06-27:
Fixed it (I guess):
- npm install node-hid
- cd node_modules/node-hid/build/HID/v0.5.1/Release/
- mkdir node-v48-linux-x64/
- cp electron-v1.1-linux-x64/* node-v48-linux-x64
- sudo npm start

stevelacy, 2016-06-28:
@chrisranderson thanks for trying that, I'll need to find a workaround for building the node and electron native hid module.
The build mismatch between electron and node is not helping...

stevelacy, 2016-06-30:
@chrisranderson  @Turbito @RicardoEPRodrigues @CompMedChem @MarcusLlewellyn 
Would you all be able to clone the latest (or git fetch) and do the following?

```
node -v
npm -v
git clone https://github.com/stevelacy/msi-keyboard-gui
rm -rf node_modules
npm i
npm run build:linux
cd dist/msi-keyboard-gui-linux-x64/
sudo ./msi-keyboard-gui
```
Thanks!

RicardoEPRodrigues, 2016-06-30:
I still have the same problem.

It just shows a blank blueish screen.

Here's the output of the instructions:
[output.txt](https://github.com/stevelacy/msi-keyboard-gui/files/341816/output.txt)

Hope it helps.

Usernameofjosh, 2016-06-30:
node -v = 6.2.2
npm -v = 3.10.2

The latest git version works, I have to set it to "normal" before the colours can be changed though
I get the blank screen on the one from https://github.com/stevelacy/msi-keyboard-gui/releases

stevelacy, 2016-06-30:
@Usernameofjosh just created a new release with the fixes: 
https://github.com/stevelacy/msi-keyboard-gui/releases/tag/0.1.1

RicardoEPRodrigues, 2016-07-01:
Sorry still black screen for me!

Tried updating node but it didn't help.

stevelacy, 2016-07-01:
@RicardoEPRodrigues Are you running it from source, or the compiled package?
What is the output from the dev console? (ctrl shift i)

RicardoEPRodrigues, 2016-07-02:
I tried the compiled package and blank screen.

Now I tried compiling from source and it worked! 😀

MarcusLlewellyn, 2016-07-03:
Sorry I haven't been able to get back to this quicker.

```
node -v : v4.47
npm -v  : 3.10.2
```

I'm still getting a blank window with both the released package, and after compiling it myself from source using the instructions above. Ctrl-Shift-i is also no longer working in either one.

RicardoEPRodrigues, 2016-07-03:
@MarcusLlewellyn I got it working by downloading the source for version 0.1.1 and compiling it. Have you tried that?

viatro, 2016-07-10:
@RicardoEPRodrigues I've tried with no success. Still the same error with mismatching paths to HID.node.
And I've managed to make it work by the method described by @chrisranderson.

ktw, 2016-07-22:
Same problem here.
Binaries does not work on openSUSE 13.2.
Source version fails and writes the following log file:
[npm-debug.log.txt](https://github.com/stevelacy/msi-keyboard-gui/files/377835/npm-debug.log.txt)

stevelacy, 2016-07-22:
@ktw unfortunately the npm debug log does not actually show the error.
Please explain the steps you took to install +run

ktw, 2016-07-22:
@stevelacy at first I tried downloading and running the binaries.

Then I followed the readme in the source:
git clone https://github.com/stevelacy/msi-keyboard-gui
cd msi-keyboard-gui
npm install
sudo npm start

stevelacy, 2016-07-22:
Try with node v6

ktw, 2016-07-22:
Now using node 6.2.2 and npm 3.10.5
[npm-debug.log.txt](https://github.com/stevelacy/msi-keyboard-gui/files/379067/npm-debug.log.txt)

stevelacy, 2016-07-22:
@ktw the npm debug output does not actually show the error.

Please run the command with which the `npm start` command will run:
```
$ npm i electron-prebuilt@1.1.0
$ ./node_modules/.bin/electron-rebuild
$ electron .
```

ktw, 2016-07-27:
Line 2 fails on a missing libusb.h file. 
I have attached the output. (Danish error messages)
[build.txt](https://github.com/stevelacy/msi-keyboard-gui/files/385827/build.txt)

stevelacy, 2016-07-27:
Please install libusb.

ktw, 2016-07-28:
Installing the source files for libusb fixed the build issue, but the binary still does not work.
(Same problem as the prebuild version)

Deminform, 2016-07-28:
Hi, You can create a short video tutorial on installing KLM for Linux/Ubuntu?

stevelacy, 2016-07-28:
@Deminform 
To busy.
Please carefully read your npm error log and search online for issues relating to npm and missing files. Update node.js, npm, and ensure you have the needed dependencies. `libusb-dev libusb-1.0-0-dev`

@ktw does `npm start` work? What does and what does not work?

Deminform, 2016-07-28:
I solved the problem!

Explain the points:
**1:** Delete the folder "msi-keyboard-gui" from the home directory (if there was one)
**2:**
         `$ git clone https://github.com/stevelacy/msi-keyboard-gui`
         `$ cd msi-keyboard-gui`
         `$ npm install`

**3:** download the archive
https://github.com/stevelacy/msi-keyboard-gui/releases/download/0.1.1/msi-keyboard-gui-linux-0.1.1.zip
**4:** unpack it to any folder and copy all the files in the folder "/home/msi-keyboard-gui/"
**5:** in the same terminal, write:
         `sudo ./msi-keyboard-gui`

**Enjoy the results! ))**

stevelacy, 2016-07-28:
@Deminform so you basically followed the install instructions.

I added the lib usb requirements to the readme. Going to close this now.

closed 2016-07-28 by stevelacy

Deminform, 2016-07-29:
How to add it in start system?

stevelacy, 2017-01-09:
Just published a release: https://github.com/stevelacy/msi-keyboard-gui/releases/tag/0.1.3
Can you confirm that the binary does run correctly, and if not the errors it produces?
Thanks