stephenlacy/msi-keyboard-gui #3: Make changes permanent

RicardoEPRodrigues, 2016-07-03:
Presently I have to open the program each time I start my pc so that the light start working.

Is it possible to make the changes permanent?

Maybe creating a thread that starts on system start-up and sets the color, then, if the user wanted to change the color, the program would only need to interact with the thread. (similar to how the apps' indicators work on Ubuntu)

stevelacy, 2016-07-03:
The program would need to start the node process with systemd, essentially meaning that there would be two processes. The GUI and a service to interact with the hid device.
Plus a file in the `~/.config` to store the last state.

I will probably do this in the future

StormBytePP, 2017-02-06:
Please, if you do a thing like this, don't force systemd, there are still many of us which hate it :)

RicardoEPRodrigues, 2017-02-06:
When this feature is done you should send it to GamingOnLinux. 😃

Mohandevir, 2017-09-17:
I'm not a programmer so it's probably not realistic, but could it  a tool to display the CLI generated by our selections? We could then add it as a script to the startup applications.

Btw, thanks for that nice tool.

Soxcks, 2019-07-12:
@Mohandevir It looks like there is a CLI version here:

https://www.npmjs.com/package/msi-keyboard-cli