stephenlacy/msi-keyboard-gui #4: Blank GUI

demaniak, 2016-08-12:
Hi.

I build from source, and tried the pre-built binaries.
The app starts up with a blank GUI (off-black background).
And does nothing else.

Node Version: v4.4.7
Ubuntu 14.04 LTS

stevelacy, 2016-08-12:
Please follow any suggestions found here; https://github.com/stevelacy/msi-keyboard-gui/issues/2

demaniak, 2016-08-18:
Sorry, been a bit pressed with work projects, I will attempts the suggestions mentioned in #2 as soon as possible (probably over the weekend). 
Thanks :)

demaniak, 2016-08-23:
Ok, looks like the issue is pretty much the same as hunted in #2 - I could not get it working, and while this looks nifty, it is unfortunately not high enough on my priority list to continue digging.
Closing this issue.

Keep up the nifty work sir, good luck.

closed 2016-08-23 by demaniak