stephenlacy/msi-keyboard-gui #6: "node-pre-gyp install --fallback-to-build" fails on Ubuntu Mate 16.04 on MSI Ghost Pro (G6QE)

raedeus, 2016-09-15:
Hello there. First of all, thanks so much for this project. I would love to alter the MSI SteelSeries keyboard in my linux build and this will be great.

After following the Dev instructions, I run into the error in the subject field:

npm ERR! Tell the auther that this fails on your system:
npm ERR!      node-pre-gyp install --fallback-to-build

I checked that both node.js and npm are installed and latest. Also libusb-dev is also installed.

Any help would be greatly appreciated.

Thanks,

Raedeus

stevelacy, 2016-09-15:
What is the npm error? The message you posted is the generic npm 'stuff broke' message.

raedeus, 2016-09-15:
Hey there. Thanks for the rapid response! I also just realized I am trying to run this for my laptops integrated SteelSeries keyboard and not the external USB keyboard. Ran the binary and just got a blank screen so perhaps this isn't the solution for me?

What is interesting is there is no actual error output. However, at the top of the process, right under "node-pre-gyp install --fallback-to-build" therre is this:

/usr/bin/env: 'node': No such file or directory

Further down into the npm ERR! messages, one states:

npm ERR! code ELIFECYCLE
npm ERR! errno ENOENT
npm ERR! syscall spawn

Is this helpful at all?

Thanks,

raedeus

stevelacy, 2016-09-15:
This has been only designed for the built-in laptop steelseries keyboard.
```
/usr/bin/env: 'node': No such file or directory
```
That is the error. Figure out what symlink name node is installed under and change it to `node`

raedeus, 2016-09-15:
Ah, I see. I would hate to have you hold my hand all the way through this, but I am a bit confused. Is "node" different than "nodejs"? I have installed the latest "nodejs" and can't seem to locate a "node" package or binary in the repos. Should I set an environment variable or symlink to /usr/bin/nodejs and call it "node" as you say?

I greatly appreciate your time, help, and patience.

Thanks,

raedeus

stevelacy, 2016-09-15:
yes
or just fix the issue: `sudo mv /usr/bin/nodejs /usr/bin/node`

raedeus, 2016-09-15:
Ah....

Simple. Elegant. Must be right. LOL.

Let me try that.

raedeus

raedeus, 2016-09-15:
Now it's complaining right off the bat:

raedeus@-----:~/build/msi-keyboard-gui$ npm install
bash: /usr/bin/npm: /usr/bin/nodejs: bad interpreter: No such file or directory

raedeus

stevelacy, 2016-09-15:
Check my updated comment, or just do `which nodejs`

raedeus, 2016-09-15:
Hello there stevelacy,

Running `which nodejs` outputs the following:

/usr/bin/nodejs

I understand this locates the binary that will be invoked, but how does this help? If I change the binary to "/usr/bin/node" the "npm install" complains right off the bat. 

As to your updated comment, I don't have any nodejs/node binaries in the "/usr/local/bin" directory so I am not sure how that will help. I am wondering if there is some dependency I am missing here, or perhaps the build needs to be updated? I see much of this project has been quiet since July. Perhaps it has to do with the 16.04 release?

Have you run a test build with your current instructions in 16.04? These recommendations you have given, graciously as they may be, haven't gotten me anywhere yet. I appreciate the help and am hoping it is simple, but so far nothing has happened. 

Perhaps I could try the CLI version....

raedeus

raedeus, 2016-09-15:
I was just reading your CLI installation notes and noticed this line:

sudo ln -s 'which nodejs' /usr/bin/node

Is this the command you are referring to? I could try that...

raedeus

raedeus, 2016-09-15:
Well, I got past the first error with:

sudo ln -s /usr/bin/nodejs /usr/bin/node

However, now there's a whole heck of a lot of errors at "sudo npm start".

Looks like make failed here:

make: *** [Release/obj.target/hidapi/hidapi/hidapi/libusb/hid.o] Error 1

I installed libusb-dev before beginning the project.

Hmmm....

raedeus

stevelacy, 2016-09-15:
Your nodejs install was setup according to the `apt` bundle, which creates a symlink to `nodejs` instead of `node`. This is due to a separate apt package called node. You now solved that issue.
This has been tested with Debian, ubuntu, 14-16.
https://github.com/stevelacy/msi-keyboard#confirmed-systems

To clean and install the required modules:

```
rm -rf node_modules
npm i
sudo npm start
```

Post the message from any errors you receive.

raedeus, 2016-09-15:
Alright. Removing the node_modules and running 'npm i' looked better, however, after 'sudo npm start' this is what I get:

`raedeus@-------:~/build/msi-keyboard-gui$ sudo npm start

> msi-keyboard-gui@0.1.1 start /home/raedeus/build/msi-keyboard-gui
> npm i electron-prebuilt@1.1.0 && ./node_modules/.bin/electron-rebuild && electron .

> electron-prebuilt@1.1.0 postinstall /home/raedeus/build/msi-keyboard-gui/node_modules/electron-prebuilt
> node install.js

msi-keyboard-gui@0.1.1 /home/raedeus/build/msi-keyboard-gui
└── electron-prebuilt@1.1.0 

npm WARN msi-keyboard-gui@0.1.1 No repository field.
npm WARN msi-keyboard-gui@0.1.1 No license field.

> node-hid@0.5.1 install /home/raedeus/build/msi-keyboard-gui/node_modules/node-hid
> node-pre-gyp install --fallback-to-build

make: Entering directory '/home/raedeus/build/msi-keyboard-gui/node_modules/node-hid/build'
  CC(target) Release/obj.target/hidapi/hidapi/libusb/hid.o
hidapi.target.mk:96: recipe for target 'Release/obj.target/hidapi/hidapi/libusb/hid.o' failed
make: Leaving directory '/home/raedeus/build/msi-keyboard-gui/node_modules/node-hid/build'
Failed to execute '/usr/bin/nodejs /home/raedeus/build/msi-keyboard-gui/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/raedeus/build/msi-keyboard-gui/node_modules/node-hid/build/HID/v0.5.1/Release/electron-v1.1-linux-x64/HID.node --module_name=HID --module_path=/home/raedeus/build/msi-keyboard-gui/node_modules/node-hid/build/HID/v0.5.1/Release/electron-v1.1-linux-x64' (1)

../hidapi/libusb/hid.c:47:20: fatal error: libusb.h: No such file or directory
compilation terminated.
make: *** [Release/obj.target/hidapi/hidapi/libusb/hid.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/raedeus/build/msi-keyboard-gui/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 4.4.0-36-generic
gyp ERR! command "/usr/bin/nodejs" "/home/raedeus/build/msi-keyboard-gui/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/raedeus/build/msi-keyboard-gui/node_modules/node-hid/build/HID/v0.5.1/Release/electron-v1.1-linux-x64/HID.node" "--module_name=HID" "--module_path=/home/raedeus/build/msi-keyboard-gui/node_modules/node-hid/build/HID/v0.5.1/Release/electron-v1.1-linux-x64"
gyp ERR! cwd /home/raedeus/build/msi-keyboard-gui/node_modules/node-hid
gyp ERR! node -v v4.2.6
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/nodejs /home/raedeus/build/msi-keyboard-gui/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/raedeus/build/msi-keyboard-gui/node_modules/node-hid/build/HID/v0.5.1/Release/electron-v1.1-linux-x64/HID.node --module_name=HID --module_path=/home/raedeus/build/msi-keyboard-gui/node_modules/node-hid/build/HID/v0.5.1/Release/electron-v1.1-linux-x64' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/raedeus/build/msi-keyboard-gui/node_modules/node-hid/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:821:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Linux 4.4.0-36-generic
node-pre-gyp ERR! command "/usr/bin/nodejs" "/home/raedeus/build/msi-keyboard-gui/node_modules/node-hid/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/raedeus/build/msi-keyboard-gui/node_modules/node-hid
node-pre-gyp ERR! node -v v4.2.6
node-pre-gyp ERR! node-pre-gyp -v v0.6.12
node-pre-gyp ERR! not ok 

npm ERR! Linux 4.4.0-36-generic
npm ERR! argv "/usr/bin/nodejs" "/home/raedeus/build/msi-keyboard-gui/node_modules/npm/bin/npm-cli.js" "rebuild" "--runtime=electron" "--target=1.1.0" "--arch=x64"
npm ERR! node v4.2.6
npm ERR! npm  v3.10.7
npm ERR! code ELIFECYCLE
npm ERR! node-hid@0.5.1 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-hid@0.5.1 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the node-hid package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs node-hid
[... 14 lines cut ...]

npm ERR! Linux 4.4.0-36-generic
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "start"
npm ERR! node v4.2.6
npm ERR! npm  v3.5.2
npm ERR! code ELIFECYCLE
npm ERR! msi-keyboard-gui@0.1.1 start: `npm i electron-prebuilt@1.1.0 && ./node_modules/.bin/electron-rebuild && electron .`
npm ERR! Exit status 255
npm ERR! 
npm ERR! Failed at the msi-keyboard-gui@0.1.1 start script 'npm i electron-prebuilt@1.1.0 && ./node_modules/.bin/electron-rebuild && electron .'.
[... cut ...]
npm ERR!     npm i electron-prebuilt@1.1.0 && ./node_modules/.bin/electron-rebuild && electron .
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs msi-keyboard-gui
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls msi-keyboard-gui
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/raedeus/build/msi-keyboard-gui/npm-debug.log
raedeus@------:~/build/msi-keyboard-gui$ 
`

stevelacy, 2016-09-15:
`
../hidapi/libusb/hid.c:47:20: fatal error: libusb.h: No such file or directory
`
Make sure that the correct libusb is installed for your OS.

raedeus, 2016-09-15:
Eureka. Looks like ubuntu has both libusb-dev and libusb-1.0-0-dev. Installed the latter and guess what, I get a nice, BLANK window.

LOL.

Same as the binaries gave me. Blank window. Looks like the right background color but I've got nothing but that. 

My proprietary nvidia drivers wouldn't have anything to do with that would they?

raedeus

raedeus, 2016-09-15:
Perhaps it's an incompatibility with compiz?

Should I switch to the Marco compositor?

raedeus

stevelacy, 2016-09-15:
Display drivers and DE have no effects. Check if there are any errors in the console. `Ctrl + Shift + i`

raedeus, 2016-09-15:
Console shows this error:

`module.js:440 Uncaught Error: Cannot find module '/home/raedeus/build/msi-keyboard-gui/node_modules/node-hid/build/HID/v0.5.1/Release/node-v48-linux-x64/HID.node'`

stevelacy, 2016-09-15:
Wait. You are running node v4.2.6?
That is the problem. Install the latest version of node (v6)
https://nodejs.org/en/download/current/

raedeus, 2016-09-15:
Hmm. I will give that a try.

raedeus

raedeus, 2016-09-16:
Hello there stevelacy,

Another step in the right direction, perhaps, but still blank screen. Perhaps I will try purging all modules and build from the beginning with the current version of node.

raedeus

raedeus, 2016-09-16:
stevelacey, thank you for all your kind attention and patience. We have solved this.

1. libusb-1.0-0-dev is needed
2. Node v6

After those two are in place, everything is great.

Thank you for your help!

raedeus

closed 2016-09-16 by raedeus

stevelacy, 2016-09-16:
Sounds good, does the project work well?