stephenlacy/msi-keyboard-gui #11: Uncaught exception loading HID.node

ohnoimdead, 2016-10-06:
module.js:440 Uncaught Error: Cannot find module '.../msi-keyboard-gui/node_modules/msi-keyboard/node_modules/node-hid/build/HID/v0.5.1/Release/node-v48-linux-x64/HID.node'

Looks like HID.node is located in ```/electron-v1.1-linux-x64``` not ```/node-v48-linux-x64```.

Ubuntu 16.04
MSI GS60 Ghost Pro

stevelacy, 2016-10-06:
What version of node do you have?
Similar issue to #2

stevelacy, 2017-01-09:
Could you do this:
```
git pull
rm -rf node_modules
npm install
sudo npm start
```

stevelacy, 2017-01-09:
Just published a release: https://github.com/stevelacy/msi-keyboard-gui/releases/tag/0.1.3
Can you confirm that the binary does run correctly, and if not the errors it produces?
Thanks

closed 2017-01-25 by stevelacy