stephenlacy/msi-keyboard-gui #14: Trying on Hackintosh OS X 10.11.6

jbmaxwell, 2016-11-14:
I disabled system integrity protection [()](www.imore.com/el-capitan-system-integrity-protection-helps-keep-malware-away) and ran from sudo, but I still get the "try from sudo" error. Any ideas what I might check?

The gui does come up, btw.

stevelacy, 2016-11-14:
This is not designed or supported for non Linux systems. The application does not find the needed mapper to the USB HID device.

jbmaxwell, 2016-11-14:
So no thoughts on where I might look to patch/hack it?

stevelacy, 2016-11-14:
You would need to get a list of the HID devices attached to the machine.
http://stackoverflow.com/questions/6477385/raw-access-to-hid-devices-in-os-x

jbmaxwell, 2016-11-15:
Of course, yes. I can look into that further... not insurmountable, by any mean... Can you point to a particular part of your code that deals with the device Ids? 
Much appreciated.

stevelacy, 2016-11-15:
The main line for retrieving and initiating the keyboard is here: https://github.com/stevelacy/msi-keyboard/blob/master/lib/findKeyboard.js#L7

And sending the feature report to the hid device: https://github.com/stevelacy/msi-keyboard/blob/master/lib/setColor.js#L24

jbmaxwell, 2016-11-15:
Right, okay. So you instantiate an HID object with the Ids... I'll see what I can do. Thanks!

stevelacy, 2017-01-09:
Quick question, could you do this:

```js
var HID = require('node-hid');
var devices = HID.devices();
console.log(devices);
```
And post the output?

jbmaxwell, 2017-01-09:
[ { vendorId: 1452,
    productId: 782,
    path: 'IOService:/AppleACPIPlatformExpert/PS2K/ps2controller/ApplePS2Controller/ApplePS2KeyboardDevice/ApplePS2Keyboard/IOHIDKeyboardDevice',
    serialNumber: '',
    manufacturer: 'Apple',
    product: 'Keyboard',
    release: 0,
    interface: -1,
    usagePage: 1,
    usage: 6 },
  { vendorId: 1452,
    productId: 613,
    path: 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC@14/XHC@14000000/HS01@14100000/Magic Trackpad 2@14100000/Actuator@2/AppleUSBTopCaseHIDDriver@14100000,2',
    serialNumber: 'CC260450KWDG61CA0',
    manufacturer: 'Apple Inc.',
    product: 'Magic Trackpad 2',
    release: 2052,
    interface: -1,
    usagePage: 65280,
    usage: 13 },
  { vendorId: 1452,
    productId: 613,
    path: 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC@14/XHC@14000000/HS01@14100000/Magic Trackpad 2@14100000/Trackpad / Boot@1/AppleUSBTopCaseHIDDriver@14100000,1',
    serialNumber: 'CC260450KWDG61CA0',
    manufacturer: 'Apple Inc.',
    product: 'Magic Trackpad 2',
    release: 2052,
    interface: -1,
    usagePage: 1,
    usage: 2 },
  { vendorId: 1452,
    productId: 613,
    path: 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC@14/XHC@14000000/HS01@14100000/Magic Trackpad 2@14100000/Accelerometer@3/AppleUSBTopCaseHIDDriver@14100000,3',
    serialNumber: 'CC260450KWDG61CA0',
    manufacturer: 'Apple Inc.',
    product: 'Magic Trackpad 2',
    release: 2052,
    interface: -1,
    usagePage: 65280,
    usage: 3 },
  { vendorId: 1452,
    productId: 782,
    path: 'IOService:/AppleACPIPlatformExpert/PS2K/ps2controller/ApplePS2Controller/ApplePS2MouseDevice/ApplePS2SmartTouchpad/IOHIDPointingDevice',
    serialNumber: '',
    manufacturer: 'Apple',
    product: 'Trackpad',
    release: 4,
    interface: -1,
    usagePage: 1,
    usage: 2 },
  { vendorId: 1452,
    productId: 613,
    path: 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC@14/XHC@14000000/HS01@14100000/Magic Trackpad 2@14100000/Device Management@0/AppleUSBTopCaseHIDDriver@14100000,0',
    serialNumber: 'CC260450KWDG61CA0',
    manufacturer: 'Apple Inc.',
    product: 'Magic Trackpad 2',
    release: 2052,
    interface: -1,
    usagePage: 65280,
    usage: 11 } ]

stevelacy, 2017-01-09:
The keyboard is not listed. What MSI laptop is this?

jbmaxwell, 2017-01-09:
Yeah, it's a Ghost Pro GS60 6QE, but I think I gave up on the backlight control when it appeared that the Hackintosh uses ApplePS2Keyboard.kext to run it... so it's not USB as far as the system is concerned. Honestly, I don't understand what's up with that. Maybe some kind of emulation?... not sure.

stevelacy, 2017-01-09:
Nice, I have that laptop as well. It sounds like apple uses a keyboard interface for all raw hid devices with a keyboard type.
If your keyboard stays on after reboot you could turn it on in linux or windows and then reboot into apple. Quite a hassle, but it might work

jbmaxwell, 2017-01-09:
hmm... can't remember, but I think it might stay on. Will check it out at some point. Thanks for looking into it.

stevelacy, 2018-04-13:
Thats good, can you write up a install/setup doc?

AutinMitra, 2018-04-17:
The gui version doesn't work with OS X (I'm looking into it), but msi-keyboard is.

stevelacy, 2018-04-17:
I'm guessing it's due to compiling for a different version of node, or node-hid mismatch.

AutinMitra, 2018-04-17:
so far, it's because of cp for mac being a bit different than it is for linux, where there is no -t option.

AutinMitra, 2018-04-17:
yup, its a node-hid mismatch

stevelacy, 2018-12-30:
Tested the latest changes on macos :smile: 
Let me know if the latest works for you, if not I'll re-open

closed 2018-12-30 by stevelacy

ErrorErrorError, 2018-12-31:
> msi-keyboard works on MacOS by using SmartTouchPad
This does not affect anything thats related to the colors since MSI has the rgb colors separated into this USB called "MSI EPF USB". Also, I used the latest release of msi-keyboard-gui but the colors don't change. The "node-hid mismatch" still happens. I will see if I can fix for macOS. I'm on node 8.14.1.

ErrorErrorError, 2018-12-31:
> node-hid mismatch" still happens

Nevermind. It does not say "node_module_mismatch" anymore, but the colors does not change.