stephenlacy/msi-keyboard-gui #17: NPM don't start

Zohran, 2016-12-14:
Hello, in first, sorry for my bad english, i'm french ! :)
I would like to use your script but he don't work. It's a log:
**0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/nodejs', '/usr/bin/npm', 'start' ]
2 info using npm@3.5.2
3 info using node@v4.2.6
4 verbose config Skipping project config: /home/zohran/.npmrc. (matches userconfig)
5 verbose stack Error: ENOENT: no such file or directory, open '/home/zohran/package.json'
5 verbose stack     at Error (native)
6 verbose cwd /home/zohran
7 error Linux 4.4.0-53-generic
8 error argv "/usr/bin/nodejs" "/usr/bin/npm" "start"
9 error node v4.2.6
10 error npm  v3.5.2
11 error path /home/zohran/package.json
12 error code ENOENT
13 error errno -2
14 error syscall open
15 error enoent ENOENT: no such file or directory, open '/home/zohran/package.json'
16 error enoent ENOENT: no such file or directory, open '/home/zohran/package.json'
16 error enoent This is most likely not a problem with npm itself
16 error enoent and is related to npm not being able to find a file.
17 verbose exit [ -2, true ]**

stevelacy, 2016-12-14:
You are in the wrong directory - use your terminal to go into the `msi-keyboard` folder, not your home folder

closed 2016-12-14 by stevelacy

Zohran, 2016-12-14:
**Sorry, i gave wrong log. When i start NPM, this is the result:**

`zohran@zohran-GS73VR-6RF:~/msi-keyboard-gui$ sudo npm start
[sudo] Mot de passe de zohran : 

> msi-keyboard-gui@0.1.1 start /home/zohran/msi-keyboard-gui
> npm i electron-prebuilt@1.1.0 && ./node_modules/.bin/electron-rebuild && electron .

> electron-prebuilt@1.1.0 postinstall /home/zohran/msi-keyboard-gui/node_modules/electron-prebuilt
> node install.js

msi-keyboard-gui@0.1.1 /home/zohran/msi-keyboard-gui
└── electron-prebuilt@1.1.0 

> node-hid@0.5.2 install /home/zohran/msi-keyboard-gui/node_modules/node-hid
> node-pre-gyp install --fallback-to-build

make : on entre dans le répertoire « /home/zohran/msi-keyboard-gui/node_modules/node-hid/build »
  CC(target) Release/obj.target/hidapi/hidapi/libusb/hid.o
hidapi.target.mk:96 : la recette pour la cible « Release/obj.target/hidapi/hidapi/libusb/hid.o » a échouée
make : on quitte le répertoire « /home/zohran/msi-keyboard-gui/node_modules/node-hid/build »
Failed to execute '/usr/bin/nodejs /home/zohran/msi-keyboard-gui/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/zohran/msi-keyboard-gui/node_modules/node-hid/build/HID/Release/HID.node --module_name=HID --module_path=/home/zohran/msi-keyboard-gui/node_modules/node-hid/build/HID/Release' (1)

node-pre-gyp ERR! Tried to download(404): https://github.com/node-hid/node-hid/releases/download/0.5.2/HID-v0.5.2-electron-v1.1-linux-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for node-hid@0.5.2 and electron@1.1.0 (electron-v1.1 ABI) (falling back to source compile with node-gyp) 
../hidapi/libusb/hid.c:47:20: fatal error: libusb.h: Aucun fichier ou dossier de ce type
compilation terminated.
make: *** [Release/obj.target/hidapi/hidapi/libusb/hid.o] Erreur 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/zohran/msi-keyboard-gui/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 4.4.0-53-generic
gyp ERR! command "/usr/bin/nodejs" "/home/zohran/msi-keyboard-gui/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/zohran/msi-keyboard-gui/node_modules/node-hid/build/HID/Release/HID.node" "--module_name=HID" "--module_path=/home/zohran/msi-keyboard-gui/node_modules/node-hid/build/HID/Release"
gyp ERR! cwd /home/zohran/msi-keyboard-gui/node_modules/node-hid
gyp ERR! node -v v4.2.6
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/nodejs /home/zohran/msi-keyboard-gui/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/zohran/msi-keyboard-gui/node_modules/node-hid/build/HID/Release/HID.node --module_name=HID --module_path=/home/zohran/msi-keyboard-gui/node_modules/node-hid/build/HID/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/zohran/msi-keyboard-gui/node_modules/node-hid/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:87:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:821:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)
node-pre-gyp ERR! System Linux 4.4.0-53-generic
node-pre-gyp ERR! command "/usr/bin/nodejs" "/home/zohran/msi-keyboard-gui/node_modules/node-hid/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/zohran/msi-keyboard-gui/node_modules/node-hid
node-pre-gyp ERR! node -v v4.2.6
node-pre-gyp ERR! node-pre-gyp -v v0.6.31
node-pre-gyp ERR! not ok 

npm ERR! Linux 4.4.0-53-generic
npm ERR! argv "/usr/bin/nodejs" "/home/zohran/msi-keyboard-gui/node_modules/npm/bin/npm-cli.js" "rebuild" "--runtime=electron" "--target=1.1.0" "--arch=x64" "--update-binary"
npm ERR! node v4.2.6
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! node-hid@0.5.2 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-hid@0.5.2 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the node-hid package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-pre-gyp install --fallback-to-build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs node-hid
[... 13 lines cut ...]
    at Process.ChildProcess._handle.onexit (internal/child_process.js:211:5)

npm ERR! Linux 4.4.0-53-generic
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "start"
npm ERR! node v4.2.6
npm ERR! npm  v3.5.2
npm ERR! code ELIFECYCLE
npm ERR! msi-keyboard-gui@0.1.1 start: `npm i electron-prebuilt@1.1.0 && ./node_modules/.bin/electron-rebuild && electron .`
npm ERR! Exit status 255
npm ERR! 
npm ERR! Failed at the msi-keyboard-gui@0.1.1 start script 'npm i electron-prebuilt@1.1.0 && ./node_modules/.bin/electron-rebuild && electron .'.
[... cut ...]
npm ERR!     npm i electron-prebuilt@1.1.0 && ./node_modules/.bin/electron-rebuild && electron .
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs msi-keyboard-gui
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls msi-keyboard-gui
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/zohran/msi-keyboard-gui/npm-debug.log
`

stevelacy, 2016-12-14:
```
../hidapi/libusb/hid.c:47:20: fatal error: libusb.h: Aucun fichier ou dossier de ce type
compilation terminated.
make: *** [Release/obj.target/hidapi/hidapi/libusb/hid.o] Erreur 1
gyp ERR! build error
```

It is explaining that you need to install libusb, such as: https://github.com/stevelacy/msi-keyboard-gui#install

Zohran, 2016-12-15:
I thought it was enough: libusb-dev. But it work when i install libusb-1.0-0-dev. Thanks ! In last, how i do for this script start in start-up of Ubuntu?

stevelacy, 2016-12-15:
http://stackoverflow.com/questions/8339555/how-to-run-a-script-at-the-start-up-of-ubuntu

Zohran, 2016-12-15:
Okay thanks ! :)
I search in documentation how i use msi-keyboard in command line but I didn't understand very well...