stephenlacy/msi-keyboard-gui #19: Problems installing on Linux mint 17.3

quixand, 2016-12-16:
Please see https://gist.github.com/quixand/20b115ba720a1464147e78d8ccf54a91
Ive already installed the nodejs-legacy package to enable the node -> nodejs symlink but I appear to be missing a file

sh: 1: ./node_modules/.bin/electron-rebuild: not found

any help appreciated

stevelacy, 2016-12-16:
Is there any error from running `npm install`?

quixand, 2016-12-16:
no not that I can tell, I did the npm install at the start of the gist

stevelacy, 2016-12-16:
What is the output of `ls node_modules/.bin`?

quixand, 2016-12-16:
nick-MSI:msi-keyboard-gui nick [master] $ ls node_modules/.bin/
babel babel-external-helpers babel-node electron electron-packager

stevelacy, 2016-12-16:
electron-rebuild was not installed, try:
```
rm -rf node_modules && npm i
sudo npm start
```
And if it is not installed try to manually install it:
``` 
npm i electron-prebuilt
```

quixand, 2016-12-16:
OK did that, electron-rebuild is there now. It's got a bit further, I had an error about libusb

```/hidapi/libusb/hid.c:47:20: fatal error: libusb.h: No such file or directory```

so a quick..
```sudo apt-get install libusb-1.0-0-dev```

got rid of that.

it seems to have completed with a few warnings  and also the following error

```
node-pre-gyp ERR! Tried to download(404): https://github.com/node-hid/node-hid/releases/download/0.5.2/HID-v0.5.2-node-v11-linux-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for node-hid@0.5.2 and node@0.10.25 (node-v11 ABI) (falling back to source compile with node-gyp) 
node-pre-gyp http 404 status code downloading tarball https://github.com/node-hid/node-hid/releases/download/0.5.2/HID-v0.5.2-node-v11-linux-x64.tar.gz 
make: Entering directory `/media/nick/msi-data/Documents/msi-keyboard-node/msi-keyboard-gui/node_modules/msi-keyboard/node_modules/node-hid/build'
 ```

I tried 
``` sudo npm start ```
The window appears but its empty?

gist of npm install and start
https://gist.github.com/quixand/d6f9732e35e60eb7fcbc7c522ffc7d0e

thanks

stevelacy, 2016-12-16:
It appears that you are using a very old version of node: v0.10.25
Please update node to v6 or v7, then re-install the packages (rebuilds node-hid)

stevelacy, 2016-12-17:
I updated the module which should help with the install

quixand, 2016-12-17:
OK got it working, wooo!
It seems that Linux mint 17.3, based on Ubuntu 14 LTS has some really old repositories for nodejs.
the default package is 

``` 
$ nodejs --version
v0.10.25
```
see also http://packages.ubuntu.com/trusty/nodejs
So I followed the instructions here https://nodejs.org/en/download/package-manager/#debian-and-ubuntu-based-linux-distributions
```
curl -sL https://deb.nodesource.com/setup_7.x | sudo -E bash -
sudo apt-get install -y nodejs
```
which installed
nodejs (7.2.1-1nodesource1~trusty1)
```
$ nodejs --version
v7.2.1
```
This also appeared to remove the legacy symlink package
```
The following packages will be REMOVED
  nodejs-dev nodejs-legacy npm
```
Thanks for your help

stevelacy, 2016-12-19:
Glad to help!

closed 2016-12-19 by stevelacy