stephenlacy/msi-keyboard-gui #20: Logo & Touchpad?

Frederick888, 2016-12-18:
Hi Steve,

First of all, thanks for great work! I've been using the CLI and I haven't changed the light color for like years lol. This is really a convenient utility.

However, I can only find the settings for left, middle and right regions, what about the logo and touchpad?

stevelacy, 2016-12-19:
I do not have access to a device with more than just the keyboard, I do not even know if those leds are controlable.

JuanCAlpizar, 2017-01-02:
@Frederick888  what keyboard do you have? 

@stevelacy if the changes I made to the CLI on its gh and can be integrated to your code I'd be glad to help in here as well for testing.

Frederick888, 2017-01-03:
Oh, sorry, I almost forgot this thread...

@JuanCAlpizar I've got a GT72S machine and now using https://github.com/bparker06/msi-keyboard to control the LEDs. And I guess that GT80 has more controllable LEDs on its lid maybe.

JuanCAlpizar, 2017-01-03:
Indeed the GT 72 and newer have more regions than the previous versions. I'm working on an update for the CLI in order to support those regions but waiting on steve confirmation on the matter; works locally for my GT72 2QE Keyboard but still needs to be merged on the msi-keyboard cli gh

stevelacy, 2017-10-17:
The core module now has more regions, I should be adding them within a month or so. (PRs accepted!)

https://github.com/stevelacy/msi-keyboard/releases/tag/msi-keyboard%400.4.0

Also adds support for a full rgb color picker!