stephenlacy/msi-keyboard-gui #21: Problem during npm install

gvader, 2016-12-29:
I am on Ubuntu 16.10 with newest node.js. Had problem with build as below:

gyp: Call to 'node -e "require('nan')"' returned exit status 127 while in binding.gyp. while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/share/node-gyp/lib/configure.js:304:16)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 4.8.0-32-generic
gyp ERR! command "/usr/bin/nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/gvader/Projects/msi-keyboard-gui/node_modules/nslog
gyp ERR! node -v v4.2.6
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok

stevelacy, 2016-12-29:
What are your versions:

```
node:
npm:
libusb:
```
I noticed this similar issue (error 127) https://github.com/nodejs/node-gyp/issues/223
I have not yet come across this error though.

gvader, 2016-12-30:
Hi @stevelacy 

Thanks for quick anwser. Versions are as below:
`
gvader@gvader-GS40-6QE-Phantom:~/Projects/msi-keyboard-gui$ dpkg -l | grep libusb
ii  libgusb2:amd64                                  0.2.9-1                                     amd64        GLib wrapper around libusb1
ii  libusb-0.1-4:amd64                              2:0.1.12-30                                 amd64        userspace USB programming library
ii  libusb-1.0-0:amd64                              2:1.0.20-1                                  amd64        userspace USB programming library
ii  libusb-dev                                      2:0.1.12-30                                 amd64        userspace USB programming library development files
ii  libusbmuxd4:amd64                               1.0.10-3                                    amd64        USB multiplexor daemon for iPhone and iPod Touch devices - library
gvader@gvader-GS40-6QE-Phantom:~/Projects/msi-keyboard-gui$ dpkg -l | grep npm
ii  node-npmlog                                     0.0.4-1                                     all          Logger with custom levels and colored output for Node.js
ii  node-read-package-json                          1.2.4-1                                     all          Read package.json for npm module for Node.js
ii  npm                                             3.5.2-0ubuntu4                              all          package manager for Node.js
gvader@gvader-GS40-6QE-Phantom:~/Projects/msi-keyboard-gui$ dpkg -l | grep nodejs
ii  nodejs                                          4.2.6~dfsg-1ubuntu5                         amd64        evented I/O for V8 javascript
ii  nodejs-dev                                      4.2.6~dfsg-1ubuntu5                         amd64        evented I/O for V8 javascript (development files)
`

stevelacy, 2016-12-30:
Please install node v6+ as that is mosylikely the problem, then re-install the project.
```
rm -rf node_modules && npm i
```
https://github.com/stevelacy/msi-keyboard-gui#nodejs-6-is-required

stevelacy, 2017-01-09:
Updates?

stevelacy, 2017-01-09:
Could you do this:
```
git pull
rm -rf node_modules
npm install
sudo npm start
```

stevelacy, 2017-01-09:
Just published a release: https://github.com/stevelacy/msi-keyboard-gui/releases/tag/0.1.3
Can you confirm that the binary does run correctly, and if not the errors it produces?
Thanks

closed 2017-01-18 by stevelacy