stephenlacy/msi-keyboard-gui #22: Create shortcut, binary, or install

sirweazel, 2017-01-03:
### how to create a shortcut that other people can use, possibly create a new binary, or a system install

I followed your instructions to download your git repository and use npm to install and a run the application. This all works perfectly and i love it. I currently am running on fedora/korora 25 and now that nodejs 6 is in the repos by default it is easy to do this. But, this is not easy for say my wife to do. So how can i install this, create a new binary to install, create a universal shortcut, or any other alternatives and ideas? Or can you update the binary downloads you currently make available?

Thanks again for your work on this, and sorry if posted this in the wrong spot.

### System versions:

```
nodejs: v6.9.2
npm: 3.10.9
libusb: 0.1.12
```

stevelacy, 2017-01-03:
Let me update the binary with the most recent updates, currently having issues with repackaging the electron app

stevelacy, 2017-01-09:
Could you do this:
```
git pull
rm -rf node_modules
npm install
sudo npm start
```

stevelacy, 2017-01-09:
Just published a release: https://github.com/stevelacy/msi-keyboard-gui/releases/tag/0.1.3
Can you confirm that the binary does run correctly, and if not the errors it produces?
Thanks

sirweazel, 2017-01-10:
Binary runs correctly using sudo.  When exiting the app, i'm not sure if if fully terminates the process though because i then have to ctrl+c on my terminal to get back to a working command prompt.  No big deal. I might try and make a policy and later try with pkexec. Do you think that is normal? I tried nohup and disown, but i'm not sure if i had the correct usage with sudo and those commands. 
Thanks again for working on this.
I assume your previous comment was for the other issue?

sirweazel, 2017-01-15:
When closing the app down, should it end all processes and return to normal bash prompt? I'm currently using sudo -b nohup to get around this issue.

stevelacy, 2017-01-16:
It should end the process when it closes, might be an electron thing.

commit 74f9eaed (2017-03-08): Close all window instances on quit - closes #22
closed 2017-03-08 by stevelacy