stephenlacy/msi-keyboard-gui #23: npm install error: Pre-built binaries not found for node-hid@0.5.2 and node@7.4.0

sergy8612, 2017-01-05:
I followed your instructions to install from development

`$ git clone https://github.com/stevelacy/msi-keyboard-gui
$ cd msi-keyboard-gui
$ npm install
`
And this is what I get after npm install:

`root@sergio-GE72-6QD:/home/sergio/msi-keyboard-gui# npm install
npm WARN deprecated babel@6.5.2: Babel's CLI commands have been moved from the babel package to the babel-cli package
npm WARN deprecated electron-prebuilt@1.4.13: electron-prebuilt has been renamed to electron. For more details, see http://electron.atom.io/blog/2016/08/16/npm-install-electron
npm WARN deprecated node-uuid@1.4.7: use uuid module instead
npm WARN prefer global node-gyp@3.4.0 should be installed with -g

> node-hid@0.5.2 install /home/sergio/msi-keyboard-gui/node_modules/node-hid
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/node-hid/node-hid/releases/download/0.5.2/HID-v0.5.2-node-v51-linux-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for node-hid@0.5.2 and node@7.4.0 (node-v51 ABI) (falling back to source compile with node-gyp) 
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/7.4.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/sergio/msi-keyboard-gui/node_modules/node-hid/.node-gyp"
make: se entra en el directorio '/home/sergio/msi-keyboard-gui/node_modules/node-hid/build'
make: *** No hay ninguna regla para construir el objetivo '../.node-gyp/7.4.0/include/node/common.gypi', necesario para 'Makefile'.  Alto.
make: se sale del directorio '/home/sergio/msi-keyboard-gui/node_modules/node-hid/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/sergio/msi-keyboard-gui/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.4.0-57-generic
gyp ERR! command "/usr/bin/nodejs" "/home/sergio/msi-keyboard-gui/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/sergio/msi-keyboard-gui/node_modules/node-hid/build/HID/Release/HID.node" "--module_name=HID" "--module_path=/home/sergio/msi-keyboard-gui/node_modules/node-hid/build/HID/Release"
gyp ERR! cwd /home/sergio/msi-keyboard-gui/node_modules/node-hid
gyp ERR! node -v v7.4.0
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/nodejs /home/sergio/msi-keyboard-gui/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/sergio/msi-keyboard-gui/node_modules/node-hid/build/HID/Release/HID.node --module_name=HID --module_path=/home/sergio/msi-keyboard-gui/node_modules/node-hid/build/HID/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/sergio/msi-keyboard-gui/node_modules/node-hid/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:885:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Linux 4.4.0-57-generic
node-pre-gyp ERR! command "/usr/bin/nodejs" "/home/sergio/msi-keyboard-gui/node_modules/node-hid/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/sergio/msi-keyboard-gui/node_modules/node-hid
node-pre-gyp ERR! node -v v7.4.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.31
node-pre-gyp ERR! not ok 
Failed to execute '/usr/bin/nodejs /home/sergio/msi-keyboard-gui/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/sergio/msi-keyboard-gui/node_modules/node-hid/build/HID/Release/HID.node --module_name=HID --module_path=/home/sergio/msi-keyboard-gui/node_modules/node-hid/build/HID/Release' (1)
msi-keyboard-gui@0.1.2 /home/sergio/msi-keyboard-gui
├── babel@6.5.2 
├─┬ babel-preset-es2015@6.18.0 
│ ├── babel-plugin-check-es2015-constants@6.8.0 
│ ├── babel-plugin-transform-es2015-arrow-functions@6.8.0 
│ ├── babel-plugin-transform-es2015-block-scoped-functions@6.8.0 
│ ├─┬ babel-plugin-transform-es2015-block-scoping@6.21.0 
│ │ ├── babel-template@6.16.0 
│ │ ├─┬ babel-traverse@6.21.0 
│ │ │ └── globals@9.14.0 
│ │ └─┬ babel-types@6.21.0 
│ │   ├── esutils@2.0.2 
│ │   └── to-fast-properties@1.0.2 
│ ├─┬ babel-plugin-transform-es2015-classes@6.18.0 
│ │ ├── babel-helper-define-map@6.18.0 
│ │ ├── babel-helper-function-name@6.18.0 
│ │ ├── babel-helper-optimise-call-expression@6.18.0 
│ │ ├── babel-helper-replace-supers@6.18.0 
│ │ └── babel-messages@6.8.0 
│ ├── babel-plugin-transform-es2015-computed-properties@6.8.0 
[... 356 lines cut ...]
npm ERR!     npm bugs node-hid
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls node-hid
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/sergio/msi-keyboard-gui/npm-debug.log
`
I attach npm-debug log file if it can help:

[npm-debug.log.txt](https://github.com/stevelacy/msi-keyboard-gui/files/688521/npm-debug.log.txt)

I have npm version 4.0.5 and nodejs 7.4.0 and ubuntu 16.04 LTS
root@sergio-GE72-6QD:/home/sergio/msi-keyboard-gui# lsb_release -a
LSB Version:	core-9.20160110ubuntu0.2-amd64:core-9.20160110ubuntu0.2-noarch:security-9.20160110ubuntu0.2-amd64:security-9.20160110ubuntu0.2-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.1 LTS
Release:	16.04
Codename:	xenial

and libusb-dev and libusb-1.0-0-dev installed

Thanks for your help,

regards

stevelacy, 2017-01-09:
```
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/7.4.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/sergio/msi-keyboard-gui/node_modules/node-hid/.node-gyp"
make: entering to directory '/home/sergio/msi-keyboard-gui/node_modules/node-hid/build'
make: *** no rule to make target '../.node-gyp/7.4.0/include/node/common.gypi', necessary for 'Makefile'. Stop.
```
This seems odd, it is trying to install as root. Can you use node v6?

stevelacy, 2017-01-09:
I just pushed changes to the build steps.
Could you do this:
```
git pull
rm -rf node_modules
npm install
sudo npm start
```

stevelacy, 2017-01-09:
Just published a release: https://github.com/stevelacy/msi-keyboard-gui/releases/tag/0.1.3
Can you confirm that the binary does run correctly, and if not the errors it produces?
Thanks

sergy8612, 2017-01-10:
Hi, I tried as you told me, but the problem persists. I don't know why node-hid is giving such problems. Maybe should I downgrade nodejs to 6.x ? 

root@sergio-GE72-6QD:/home/sergio/msi-keyboard-gui# git pull
remote: Counting objects: 15, done.
remote: Compressing objects: 100% (15/15), done.
remote: Total 15 (delta 5), reused 0 (delta 0), pack-reused 0
Unpacking objects: 100% (15/15), done.
De https://github.com/stevelacy/msi-keyboard-gui
   fc47d0e..77ecb23  master     -> origin/master
 * [new tag]         0.1.3      -> 0.1.3
Updating fc47d0e..77ecb23
Fast-forward
 README.md    |  4 ++--
 index.js     |  2 +-
 package.json | 26 +++++++++++++-------------
 src/index.js |  2 --
 4 files changed, 16 insertions(+), 18 deletions(-)
root@sergio-GE72-6QD:/home/sergio/msi-keyboard-gui# ls
assets    LICENSE       npm-debug.log      package.json  src
index.js  node_modules  npm-debug.log.txt  README.md     SUPPORTED.md
root@sergio-GE72-6QD:/home/sergio/msi-keyboard-gui# rm -fr node_modules
root@sergio-GE72-6QD:/home/sergio/msi-keyboard-gui# npm install
npm WARN deprecated node-uuid@1.4.7: use uuid module instead
npm WARN prefer global node-gyp@3.5.0 should be installed with -g

> node-hid@0.5.2 install /home/sergio/msi-keyboard-gui/node_modules/node-hid
> node-pre-gyp install --fallback-to-build

node-pre-gyp ERR! Tried to download(404): https://github.com/node-hid/node-hid/releases/download/0.5.2/HID-v0.5.2-node-v51-linux-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for node-hid@0.5.2 and node@7.4.0 (node-v51 ABI) (falling back to source compile with node-gyp) 
gyp WARN EACCES user "root" does not have permission to access the dev dir "/root/.node-gyp/7.4.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/sergio/msi-keyboard-gui/node_modules/node-hid/.node-gyp"
make: entering directory '/home/sergio/msi-keyboard-gui/node_modules/node-hid/build'
make: *** No rule to make target '../.node-gyp/7.4.0/include/node/common.gypi', necessary for 'Makefile'.  Stop.
make: exiting directory '/home/sergio/msi-keyboard-gui/node_modules/node-hid/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/home/sergio/msi-keyboard-gui/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.4.0-57-generic
gyp ERR! command "/usr/bin/nodejs" "/home/sergio/msi-keyboard-gui/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/sergio/msi-keyboard-gui/node_modules/node-hid/build/HID/Release/HID.node" "--module_name=HID" "--module_path=/home/sergio/msi-keyboard-gui/node_modules/node-hid/build/HID/Release"
gyp ERR! cwd /home/sergio/msi-keyboard-gui/node_modules/node-hid
gyp ERR! node -v v7.4.0
gyp ERR! node-gyp -v v3.5.0
gyp ERR! not ok 
node-pre-gyp ERR! build error 
node-pre-gyp ERR! stack Error: Failed to execute '/usr/bin/nodejs /home/sergio/msi-keyboard-gui/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/sergio/msi-keyboard-gui/node_modules/node-hid/build/HID/Release/HID.node --module_name=HID --module_path=/home/sergio/msi-keyboard-gui/node_modules/node-hid/build/HID/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/sergio/msi-keyboard-gui/node_modules/node-hid/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
node-pre-gyp ERR! stack     at emitTwo (events.js:106:13)
node-pre-gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
node-pre-gyp ERR! stack     at maybeClose (internal/child_process.js:885:16)
node-pre-gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
node-pre-gyp ERR! System Linux 4.4.0-57-generic
node-pre-gyp ERR! command "/usr/bin/nodejs" "/home/sergio/msi-keyboard-gui/node_modules/node-hid/node_modules/.bin/node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd /home/sergio/msi-keyboard-gui/node_modules/node-hid
node-pre-gyp ERR! node -v v7.4.0
node-pre-gyp ERR! node-pre-gyp -v v0.6.31
node-pre-gyp ERR! not ok 
Failed to execute '/usr/bin/nodejs /home/sergio/msi-keyboard-gui/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/sergio/msi-keyboard-gui/node_modules/node-hid/build/HID/Release/HID.node --module_name=HID --module_path=/home/sergio/msi-keyboard-gui/node_modules/node-hid/build/HID/Release' (1)
msi-keyboard-gui@0.1.3 /home/sergio/msi-keyboard-gui
..... 

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@^1.0.0 (node_modules/chokidar/node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@1.0.17: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})
npm ERR! Linux 4.4.0-57-generic
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! node v7.4.0
npm ERR! npm  v4.0.5
npm ERR! code ELIFECYCLE

npm ERR! node-hid@0.5.2 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-hid@0.5.2 install script 'node-pre-gyp install --fallback-to-build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the node-hid package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
[... cut ...] -build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs node-hid
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls node-hid
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/sergio/msi-keyboard-gui/npm-debug.log
[npm-debug.log.txt](https://github.com/stevelacy/msi-keyboard-gui/files/697348/npm-debug.log.txt)

stevelacy, 2017-01-10:
Did you run the binary? That does not need to be installed

sergy8612, 2017-01-10:
Yes! I just tried running the binary.

And it works well!!!

Thank you very much!!

closed 2017-01-10 by stevelacy