stephenlacy/msi-keyboard-gui #24: blank window when i run the gui

pariosa, 2017-01-09:
### Issue

When i run npm start i get an electron window, but it is completely empty 

http://i.imgur.com/UWwLMpc.png

### System versions: Ubuntu 16

```
nodejs:v6.9.4
npm:3.10.10
libusb:2:0.1.12-28
```

pariosa, 2017-01-09:
[msi-keyboard-gui] npm start                                            master 

> msi-keyboard-gui@0.1.2 start /home/[user]/msi-keyboard-gui
> electron-rebuild && electron .

✔ Rebuild Complete
^C
[msi-keyboard-gui] node -v                                              master 
v6.9.4
[msi-keyboard-gui] npm -v                                               master 
3.10.10

stevelacy, 2017-01-09:
Thank you - this issue has been reported already: #2 #4 #5 #10 

What is the error in the dev console? <kbd>CTRL</kbd>+<kbd>Snift</kbd>+<kbd>i</kbd>
The error is node-hid's native module not getting loaded for electron. All the electron rebuild steps sometimes manage to get the ENV correct and build the native module.

pariosa, 2017-01-09:
Uncaught Error: Cannot find module '/home/[user]/msi-keyboard-gui/node_modules/node-hid/build/HID/Release/HID.node'

closed 2017-01-09 by pariosa

stevelacy, 2017-01-09:
Could you do this:
```
git pull
rm -rf node_modules
npm install
sudo npm start
```

I pushed changes which should fix the need to monkey patch the build

pariosa, 2017-01-09:
that worked!