stephenlacy/msi-keyboard-gui #25: Tornado F5 (MSI 16L13) not working

jonathanfvmtl, 2017-01-13:
### Issue

Hi! I tried a bunch of times to use msi-keyboard command and gui, but it doesn't seem to work on my laptop (I have a barebones MSI model 16L13, which I believe is the base of a MSI GT63VR). How can I do further testing to see if it would be possible to make it work? Thank you! I'm pretty sure I do have a SteelSeries keyboard, as it looks the same as any other SS keyboards I've seen pictures of.

### System versions:

```
nodejs: 6.9.4
npm: 3.10.10
libusb: 2.1.0.20-1
```

stevelacy, 2017-01-13:
Can you post the output of this:

```js
var HID = require('node-hid');
var devices = HID.devices();
console.log(devices);
```

jonathanfvmtl, 2017-01-13:
Here's the result:

[ { vendorId: 6000,
    productId: 65280,
    path: '0001:0002:00',
    serialNumber: 'MSI EPF USB',
    manufacturer: 'MSI EPF USB',
    product: 'MSI EPF USB',
    release: 272,
    interface: 0 },
  { vendorId: 1133,
    productId: 49257,
    path: '0001:0004:00',
    manufacturer: 'Logitech',
    product: 'USB Laser Mouse',
    release: 22017,
    interface: 0 } ]

stevelacy, 2017-01-13:
That is fine. What happens when you try to use the GUI, are there any errors in the dev console? <kbd>CTRL</kbd>+<kbd>Snift</kbd>+<kbd>i</kbd>

jonathanfvmtl, 2017-01-13:
All the console shows is this:

/home/mrx/msi-klm/msi-keyboard-gui/dist/app/node_modules/react-dom/lib/ReactDOM.js:73

Download the React DevTools and use an HTTP server (instead of a file: URL) for a better development experience: https://fb.me/react-devtools

stevelacy, 2017-01-13:
What is the error causing the program to not work, does it show the controls?

jonathanfvmtl, 2017-01-13:
The program controls show properly, everything looks like it should work, but nothing happens, my keyboard doesn't light up. Same when using the command line. No errors outputted in the shell, but nothing happens either.

stevelacy, 2017-01-13:
Did you turn up your keyboard LED hardware switch? `fn + NUMPAD_MINUS`

jonathanfvmtl, 2017-01-13:
Yes. But it never turned the LEDs on to start with. Is it possible that it's my keyboard that's defective? I guess I should install Windows to test it out.

stevelacy, 2017-01-13:
If it did not turn them red (default) then there may be an issue with the keyboard, or your BIOS. I am running windows/linux with legacy, non UEFI.

jonathanfvmtl, 2017-01-13:
Yes, I am also running Linux with legacy. But I'm not running Windows. I'll see if I can make a USB stick with Windows on it and boot on that, see if I can get the LEDs to light up.

Also, I think my BIOS isn't a stock BIOS from MSI, but a custom BIOS from Eurocom (different motherboard and desktop components).

stevelacy, 2017-01-13:
Yeah, there should not be any reason why it won't work. Let me know how it goes. Also try restarting to your BIOS and seeing if you can turn on the leds while it is in boot-mode. On an earlier MSI (maybe bios) you needed to enable the lights in windows first, then reboot to linux.
I have not had that issue with the ghost 60 or the gt60.

jonathanfvmtl, 2017-01-13:
I already checked all the BIOS options (first thing I did!), but there was nothing for the LEDs or keyboard backlight.

But enabling the LEDs in Windows first would make a lot of sense. I don't have time to do it right now, but I'll make a Windows Live USB either later today or sometimes during the weekend and report back.

jonathanfvmtl, 2017-01-16:
Just wanted to reassure you I haven't forgotten, I've just been very busy. But I still intend to let you know how it goes.

jonathanfvmtl, 2017-01-16:
Alright, I ended up just installing Windows to see if it would work. It didn't. You were right, my keyboard's backlight is broken. Thanks for your help, and sorry for taking your time when the solution was so obvious.

closed 2017-01-16 by jonathanfvmtl

stevelacy, 2017-01-16:
No worries, I hope you can get a RMA.

jonathanfvmtl, 2017-01-17:
Yes, I'm sure I can. Not sure if it's worth it tho.