stephenlacy/msi-keyboard-gui #29: cannot open device with vendor id 0x1770 and product id 0xff00

madddmike, 2017-06-16:
### Issue

Trying to run msi-keyboard-gui from 1.4 pre-built binaries on an MSI GE60 laptop(model: MS-16GC) and receiving the following error even when running using sudo:

cannot open device with vendor id 0x1770 and product id 0xff00

### System versions:  Manjaro Linux

```
nodejs: 7.10.0-2 
npm: 4.6.1-1 
libusb: 1.0.21-1
```

stevelacy, 2017-06-16:
did you install libusb-dev?

Darkcast, 2017-07-18:
Hi, I have a MSI VR Ready GS43VR Phantom Pro-006 running kali linux and I have all of the requirements installed and I run it as root and get the same error. has anyone found an fix for this ?

stevelacy, 2017-07-18:
I seriously hope that kali is not your main OS. if it is.... yikes...

Did you install libusb-dev?

Darkcast, 2017-07-18:
yes I have libusb-dev? installed

stevelacy, 2017-07-18:
can you install and run the core library and example?
https://github.com/stevelacy/msi-keyboard
https://github.com/stevelacy/msi-keyboard/blob/master/examples/color.js

closed 2017-11-03 by stevelacy