stephenlacy/msi-keyboard-gui #30: Window appears but nothing's inside

Tarentaal, 2017-06-24:
### Issue

I have managed to install everything, but I am left with one problem: the application window opens, but it's empty - there are no colours to select or anything else in it.

I appreciate your effort with this utility and would really like to be able to use it. Is there anything I'm doing wrong?

### System versions:

Linux Mint 18.1

```
nodejs:
npm:
libusb:
```

stevelacy, 2017-06-26:
#24 #15 #10 #5 #2 

The system versions are required for me to identify any issues. You have a node version mismatch, did you try the prebuilt binaries?

Tarentaal, 2017-06-29:
Sorry for not replying sooner, my internet broke down.
Ok, I think I was able to install the prebuilt binaries. My node version is now 8.1.2 and my npm version is 5.0.3. Now when I run the sudo npm start command it stall when it hits these lines:
```
src/lib/keyboard.js -> dist/app/src/lib/keyboard.js
src/views/App.js -> dist/app/src/views/App.js
src/views/ColorPicker.js -> dist/app/src/views/ColorPicker.js
src/views/Keyboard.js -> dist/app/src/views/Keyboard.js
src/views/Selector.js -> dist/app/src/views/Selector.js`
```
And no window appears. I am sorry if I am doing something stupid:(

stevelacy, 2017-06-29:
Delete everything, install from the latest prebuilt binary release: https://github.com/stevelacy/msi-keyboard-gui/releases/tag/0.1.4

Tarentaal, 2017-06-29:
Thanks, will do!

Tarentaal, 2017-06-30:
Still not working... but I'm not able to uninstall node and npm completely. Do you know how I can completely remove them from Linux so that I can do a fresh install?

Tarentaal, 2017-06-30:
Not sure what I did, but I got it working! Thank you for all your help. Thanks for your app also it makes going over to Linux much easier on the eyes!

stevelacy, 2017-06-30:
Good to know! Cheers

closed 2017-06-30 by stevelacy