stephenlacy/msi-keyboard-gui #31: Unsupported keyboard?

knuxyl, 2017-07-14:
I have a Steelseries Apex 100 keyboard. Everything directs me to this application for changing the colors. I actually want to turn them off and I don't want to take out all 352 screw to cut the power. 

I noticed in findKeyboard.js it is set to only a specific Vendor and ID number. I changed them to the ones my keyboard is using (hex to decimal) and got unable to get feature report error.

This keyboard uses the Steelseries Engine 3. Wine will not load the windows engine and they don't make one for linux so this is as close as I can get to getting rod of this blinding light with software.

Btw, it's a pain to get the matching npm and nodejs version required for use on Ubuntu 16.04. I couldn't even use the standard nodejs repo because of npm problems.

stevelacy, 2017-07-15:
This is for the internal stealseries laptop keyboards, not an external USB device. That would require a different USB serial port interface.
If the windows steelseries engine 3 has support for both the USB and HID internal devices then it is possible to reverse engineer the USB interface. I would verify the vendor and product id just to ensure it is correct, and use sudo.

I've installed the correct node and npm versions a few times on ubuntu server 16.04.
Install using `n` https://github.com/tj/n `sudo n 6`

To start determining the feature reports, please get the `data` and or `error` events from `node-hid`: https://github.com/node-hid/node-hid#device--new-hidhidvidpid

stevelacy, 2017-07-15:
```js
var HID = require('node-hid');
var device = new HID.HID(vid, pid);
device.on('data', console.log);
device.on('error', console.log);
device.close();
```

knuxyl, 2017-07-15:
The closest I can get to what language that is is PHP :p What is that?

stevelacy, 2017-07-15:
save that to a javascript file and run it with node.

`sudo node file.js`

closed 2017-10-17 by stevelacy