stephenlacy/msi-keyboard-gui #34: Simple mod for Windows 10?

Hvass-Labs, 2017-11-22:
I have an MSI GT62VR with a SteelSeries RGB keyboard. I use it with Windows 10 and Ubuntu 16. On Windows I have a problem when the laptop is running in Eco power-saving-mode on battery. Sometimes when the machine is under load it will not immediately register a key is released, so it writes things like "hellooooooooooooo" as if the o-key was stuck.

Neither MSI nor SteelSeries have provided any support on this problem. I have already tried updating all relevant drivers etc.

I think the problem might be the SteelSeries Engine that is trying to intercept macro keys and what-not, but has not been optimized properly. I've tried to increase the thread priority and also just killing the SteelSeries Engine thread. Although it'll take longer to determine if that is indeed the problem, because the problem only surfaces occasionally.

This problem only happens in Windows 10 and not in Ubuntu.

If the machine was first booted in Windows, then I can press the SteelSeries logo-key a few times and I get the backlit RGB keys in Linux. If the machine has not previously been booted in Windows, then the backlight is just white in Linux.

Two questions:

1) Do you think the SteelSeries Engine is a likely explanation for the problem I am observing with stuck keys, or do you have another suggestion for what I could try, given your knowledge of these matters?

2) Would it be possible to mod your code so it runs on Windows 10 and merely turns on the RGB backlight? I don't need any control or macros or anything, I just want the RGB light to turn on. Or is there another simple way of doing this?

Thanks!

stevelacy, 2017-11-22:
The stuck keys is a problem with windows, and or other software, there are hundreds of stuck key reports online. I have never used the steelseries engine in windows beyond the first version of a few years ago.

If you can do this and post the output I'll know if it's possible:
`$ npm i node-hid`
```js
var HID = require('node-hid');
var devices = HID.devices();
console.log(devices);
```

Hvass-Labs, 2017-11-23:
Thanks for the quick reply.

It has probably been 15 years since I've worked with java / javascript. But I managed to install npm and node and put the code in foo.js and ran that file with node on Ubuntu. This is the output I got.

	[ { vendorId: 6000,
	    productId: 65280,
	    path: '0001:0004:00',
	    release: 272,
	    interface: 0 },
	  { vendorId: 7805,
	    productId: 11502,
	    path: '0001:0003:00',
	    release: 256,
	    interface: 0 },
	  { vendorId: 7805,
	    productId: 11502,
	    path: '0001:0003:01',
	    release: 256,
	    interface: 1 } ]

stevelacy, 2017-11-23:
Thanks, Interesting:
```
{ vendorId: 6000,
    productId: 65280,
    path: '0001:0004:00',
    release: 272,
    interface: 0 }
```
That is the correct vendor object.
Can you try this?

```
$ git clone https://github.com/stevelacy/msi-keyboard
$ cd msi-keyboard
$ npm i
$ node ./examples/color.js
```

Hvass-Labs, 2017-11-24:
I tried running those commands in Ubuntu and got the following error message:

	magnus@torpedo:~/msi-keyboard$ node ./examples/color.js
	/home/magnus/msi-keyboard/node_modules/node-hid/nodehid.js:24
	    this._raw = new (Function.prototype.bind.apply(binding.HID,
		        ^

	Error: cannot open device with vendor id 0x1770 and product id 0xff00
	    at Error (native)
	    at new HID (/home/magnus/msi-keyboard/node_modules/node-hid/nodehid.js:24:17)
	    at module.exports (/home/magnus/msi-keyboard/lib/findKeyboard.js:8:14)
	    at Object.<anonymous> (/home/magnus/msi-keyboard/examples/color.js:1:92)
	    at Module._compile (module.js:410:26)
	    at Object.Module._extensions..js (module.js:417:10)
	    at Module.load (module.js:344:32)
	    at Function.Module._load (module.js:301:12)
	    at Function.Module.runMain (module.js:442:10)
	    at startup (node.js:136:18)

stevelacy, 2017-11-24:
Try to open your cmd or shell as admin, probably right click and "run as admin"

Hvass-Labs, 2017-11-25:
I'm running this on Linux so far because I don't have any dev-tools installed on Windows (git, java, etc.) 

The following command worked and turned on the keyboard's RGB lights on Ubuntu:

    $ sudo node ./examples/color.js 

Questions:

1) How can I put this command in a script so every time Ubuntu wakes up from sleep this command is run? Because the keyboard lights are turned off when it wakes up. How do you do it?

2) What is the minimal requirements for getting this working on Windows 10? Do I need to install git, java, npm, etc.? Or can I just copy a few of the .js files? I realize you're not a Windows user, but any idea how many dependencies etc. there are in your code and how much I would need to get this working on Windows?

stevelacy, 2017-11-27:
startup scripts https://www.cyberciti.biz/tips/linux-how-to-run-a-command-when-boots-up.html

There is no telling what is needed on windows, if you can run that command on windows and get an output then it might work.

Hvass-Labs, 2017-11-29:
Thanks very much for your help I appreciate it!

closed 2017-11-29 by Hvass-Labs

bvobart, 2017-11-29:
Since I'm running a dual boot with Windows 10 x64, I thought I'd give this a shot as soon as I'd boot into Windows again. Even though the issue was closed before I could, I wanted to try and share my findings anyway, mainly since I'm not completely fond of the SteelSeries Engine either. 

Your first code snippet with node-hid:
```js
var HID = require('node-hid');
var devices = HID.devices();
console.log(devices);
```
printed a pretty large list, but I did find the device of interest:
```
{ vendorId: 6000,
    productId: 65280,
    path: '\\\\?\\hid#vid_1770&pid_ff00#7&2b03053&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}',
    serialNumber: 'MSI EPF USB',
    manufacturer: 'MSI EPF USB',
    product: 'MSI EPF USB',
    release: 272,
    interface: -1,
    usagePage: 65440,
    usage: 1 },
```

Then cloning the `msi-keyboard` repository and running `node ./examples/color.js` worked as well, though instead of the three colours that it displays while I'm in Ubuntu, it displays one solid, though slightly green/blue-ish white colour. Why? I have no idea... The wave examples do work as they should though.

Finally, I also tried running `msi-keyboard-gui` on Windows. I removed the `os` tag from the package.json and was then able to `npm install` it. Then running `npm start` started the building process, which finished quickly enough, but it took quite some time for the application window to actually appear. Once it did though, the application launched correctly. I did a quick run-through of the colours and modes and they all seemed to work perfectly as well 😃 @stevelacy It might be easier to get this working properly on Windows than you previously thought 😋

So, as an answer to question nr. 1, @Hvass-Labs, in order to run from source like I just did, you'll need to have NodeJS and npm installed. Git is not strictly required since it is also possible to download the repository's contents as a .zip from GitHub itself, but it is useful to have anyways.

stevelacy, 2017-11-29:
WAT WAT WAT OMG.
Ok amazing, did it only take `npm start` to build correctly? If so I should just be able to alter the build command and add a `build:windows`

stevelacy, 2017-11-29:
@bvobart I'll do a windows dual boot and install the needed dev tools to try it there, thanks

bvobart, 2017-11-30:
Yeah, I was surprised that it worked that easily as well :P Luckily. this is NodeJS and ReactJS, which are mostly designed to be OS independent anyway. It was indeed only `npm start` (and thus `build:app`) that made the app work properly. The only problem I noticed was that it took quite some time (+- a minute at first launch or so) for the app to actually launch, although I have no idea why. Pre-built binaries would probably launch much quicker, though I'm not sure about that, as I have no experience building Electron apps and the like. Let me know once you've implemented a `build:windows` command, then I'll go and test it as well ;)

Hvass-Labs, 2017-11-30:
@bvobart Thanks very much!

stevelacy, 2017-12-05:
check this out:
[image]

windows lol

Windows sucks lol...
https://github.com/stevelacy/msi-keyboard-gui/commit/fadafa2a568011dfac591899397e9458583adf32#diff-b9cfc7f2cdf78a7f4b91a753d10865a2

stevelacy, 2017-12-05:
Just released a build for windows: https://github.com/stevelacy/msi-keyboard-gui/releases/tag/0.2.1
@bvobart @Hvass-Labs @knuxyl

Hvass-Labs, 2017-12-05:
Thanks!

I actually prefer Windows and am not too happy with Linux nowadays. I used it 20 years ago and I think it has gotten worse since then. I'm now having massive problems with WiFi drivers, for example, and I can see that many others have the same problems, but it doesn't get fixed. It also took me a week to make a full install of Linux to do Deep Learning, because Ubuntu was so incredibly fragile if I installed something it didn't like, then I would have to reinstall everything. The problem I'm having with stuck keys in Windows, I don't think it's a Windows problem as much as a driver problem, probably with the SteelSeries Engine. So hopefully your tools may help with that. Thanks again!

bvobart, 2017-12-05:
Nice!! Just downloaded and launched the new release and it seems to work properly! I haven't launched the app from source as of yet, but it's great to see that this app now actually has a fully working release on Windows.

Personally, I prefer Windows for stability, everyday use and gaming, but Linux for development purposes. I don't have any problems with WiFi drivers and the like, although I do apparently have problems with graphics drivers (or Xorg), seeing as it seemingly switches over to software based rendering instead of hardware-accelerated rendering, which causes scrolling, moving / resizing windows, etc. to become incredibly slow. A reboot fixes it though. 

Anyway, I hope this app (now with Windows support) will help you in your quest to fix those stuck keys! ;) As soon as I have some time for it again, I'll see if I can add some features here and there.