stephenlacy/msi-keyboard-gui #35: Blank window when running from source on NodeJS v8

bvobart, 2017-11-28:
### Issue

Really nice and simple app, I love it. It works perfectly when I install using the prebuilt binaries, but when installing from source, then the app launches with just a blank window, only showing the correct background colour. When I open the developer console, it shows a NODE_MODULE_VERSION mismatch of a node in `node-hid`:

```
/home/bart/Documents/msi-keyboard-gui/dist/app/node_modules/bindings/bindings.js:88 

Uncaught Error: The module '/home/bart/Documents/msi-keyboard-gui/dist/app/node_modules/node-hid/build/Release/HID.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 50. This version of Node.js requires
NODE_MODULE_VERSION 54. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or`npm install`).
    at process.module.(anonymous function) [as dlopen] (ELECTRON_ASAR.js:173:20)
    at Object.Module._extensions..node (module.js:598:18)
    at Object.module.(anonymous function) [as .node] (ELECTRON_ASAR.js:173:20)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at bindings (/home/bart/Documents/msi-keyboard-gui/dist/app/node_modules/bindings/bindings.js:81:44)
    at Object.<anonymous> (/home/bart/Documents/msi-keyboard-gui/dist/app/node_modules/node-hid/nodehid.js:5:34)
```

I've tried updating the `msi-keyboard`, and Electron dependencies to their latest versions in the package.json, then removing the `node_modules` and `dist` folders, and then reinstalling using `npm install`. This still resulted in the same error.

As suggested [here](https://github.com/node-hid/node-hid/issues/209), I've also tried `npm rebuild node-hid --force` to no avail. Adding `electron-rebuild --force` to the package.json as a script and then executing that does not solve the problem either...

### System versions:
As for the the libusb version, that is what `dpkg -l | grep libusb-dev` returned. 
```
nodejs: 8.9.1
npm: 5.5.1
libusb: 2:0.1.12-28
```

stevelacy, 2017-11-28:
This is a systematical issue relating to electron, node, and native modules. [prior issues](https://github.com/stevelacy/msi-keyboard-gui/issues?utf8=%E2%9C%93&q=is%3Aissue+is%3Aclosed+blank+)
For the successful build on the releases I had to manually set the node version to match the electron core v8 version and run the build.
I have tried several variations of electron rebuild without success, even resetting the entire env with electron's versions still builds a package depending on a different version of node v8.
I will push a commit shortly that may fix the issue as it updates dependencies and partially alters the build command.

stevelacy, 2017-11-28:
Try now with latest on master

bvobart, 2017-11-28:
Thanks for your quick response! I figured that it would indeed be an issue with the dependency on an older version of Electron, With your latest version on the master (v0.2.1), the app starts up correctly and works as it should. I'll update `SUPPORTED.md` with my information later tonight. Thanks!

closed 2017-11-28 by bvobart