stephenlacy/msi-keyboard-gui #39: Auto startup?

natedawwgg, 2018-03-09:
### Issue
Is there auto startup?
**Describe your issue, comment or suggestion**
I have everything working perfectly fine, but once reset or sleep the keyboard defaults to standard. Is there any work around to this at all?

Thank you again for all of this!

stevelacy, 2018-03-14:
It would be easier to get https://github.com/stevelacy/msi-keyboard to run at startup with a preset color scheme.
https://askubuntu.com/questions/228304/how-do-i-run-a-script-at-start-up

I will eventually add persistence to this project which will enable it to remember the last setting upon start, and thus just starting the application on startup will re-enable the keyboard colors. #3

thereelists, 2018-04-06:
Same question for Windows 10, I've got it in Startup and windows pops up the window, can I get it to apply a preset color scheme at startup without seeing the window? Like with regedit  -hidewnd or something?

stevelacy, 2018-09-13:
Eventually, I don't use my MSI much anymore, but for startup I was considering linux first. Not sure what would be needed for full windows startup support.

But config saving is the first item needed, then re-syncing with the config and applying it on startup.

HerMajestyDrMona, 2018-09-13:
For autostart on Windows - I simply created a task in the Task Scheduler, triggered on "On workstation unlock of any user" event. Also changed "Conditions" to work on battery too. Whenever I log in it starts msi-keyboard-gui.exe.
I think autostart isn't that big problem as auto-applying settings. If it worked, we could simply create a VBScript which could run it hidden (without displaying a window), a perhaps quit after few seconds.

stevelacy, 2018-09-13:
if you can make it so checking a box will auto-enable/disable that on windows and linux I will merge it

HerMajestyDrMona, 2018-09-13:
I'm sorry, but I'm worried I'll be unable to do it with Node.js any soon (need to learn it, and soon sending laptop for repairs) :(
Maybe it would be faster if I wrote a tutorial how to do this, on Wiki page?

NikolaasBender, 2018-12-04:
uhh, for the startup thing, could you just have a little script that runs at startup. That script runs the app without any graphics then exits the app. saving the colors you can just use a text file with hex color numbers or predetermined numbers to denote each color. overwrite the file every time you save and read it on every startup. Idk how file IO is on node.js but i asume its not super complicated. the hardest part will probably be startup script but bash is pretty straightforward and powerful.
just sharing ideas, not telling you how to build your code.

NikolaasBender, 2018-12-05:
i'm going to see if this works, but on linux I think something like this will work to make it run at startup
"sudo ./Downloads/msi-keyboard-gui-linux-x64/msi-keyboard-gui && pkill msi-keybaord-gui"

rijnhard, 2019-05-09:
FYI: this alternative has a nice way of handling the autostart https://github.com/Gibtnix/MSIKLM

You may want to copy that udev handling in linux, just give attribution to the original author for that. But thats pretty sweet because it only fires if the device is detected.