stephenlacy/msi-keyboard-gui #50: cannot open device with vendor id 0x1770 and product id 0xff00

myrrdin1975, 2018-06-21:
### Issue

I cannot get the program to work, everytime I try (even when running as root) this is the error I get: 
cannot open device with vendor id 0x1770 and product id 0xff00

### System versions:
Kubuntu 18.04
```
nodejs:
npm:
libusb:
```

stevelacy, 2018-06-21:
I need more information to help, you basically said "I have a problem"

myrrdin1975, 2018-06-21:
so admittedly I don't know enough to know what information you need or how to get it :(

stevelacy, 2018-06-21:
The information is listed at the bottom of the issue template,

```
node version:
npm version:
libusb version:
machine name/model:
```

myrrdin1975, 2018-06-21:
ok, so it's an MSI GE70 2PE Apache Pro
The other stuff I'm not sure how to get the version numbers. Can I run a terminal command to get them?

stevelacy, 2018-06-21:
yes

myrrdin1975, 2018-06-21:
so, if I'm right it's as follows:
node 3.6.2-1
npm 3.5.2
libusb kept saying command not found when trying to us libusb --version

stevelacy, 2018-06-21:
node 6+ is required, you have 3.6
https://github.com/stevelacy/msi-keyboard-gui#nodejs-6-is-required

myrrdin1975, 2018-06-22:
so I no have:
node - 8.11.3
npm - 5.6.0
libusb 2:1.0.21-2

stevelacy, 2018-06-22:
Can you install the base library and check to see if you can run the `colors.js` file with `node colors.js`?

https://github.com/stevelacy/msi-keyboard
https://github.com/stevelacy/msi-keyboard/blob/master/examples/color.js

```
git clone https://github.com/stevelacy/msi-keyboard
cd msi-keyboard
npm install
sudo node examples/color.js
```

myrrdin1975, 2018-06-22:
So, please forgive my ignorance but I have no idea what you mean by 'base library'

stevelacy, 2018-06-23:
the commands I posted in the bottom of the comment show how to install the base library, the library that runs behind the GUI

myrrdin1975, 2018-06-23:
Ok, so this is going to be a lot... The first command worked great, but the rest had issues. Also, for whatever reason, I couldn't get the cd to go to where I needed it, so I found the file and opened a terminal there to do the rest of the commands in sudo.
Below is the result:

First Command:
ken@ken-GE70-2PE:~$ sudo -i
[sudo] password for ken: 
root@ken-GE70-2PE:~# git clone https://github.com/stevelacy/msi-keyboard
Cloning into 'msi-keyboard'...
remote: Counting objects: 263, done.
remote: Total 263 (delta 0), reused 0 (delta 0), pack-reused 263
Receiving objects: 100% (263/263), 43.03 KiB | 6.15 MiB/s, done.
Resolving deltas: 100% (149/149), done.
root@ken-GE70-2PE:~# cd msi-keyoard
-bash: cd: msi-keyoard: No such file or directory
root@ken-GE70-2PE:~# cd ./msi-keyboard-gui-linux-x64
-bash: cd: ./msi-keyboard-gui-linux-x64: No such file or directory
root@ken-GE70-2PE:~# list

Command 'list' not found, did you mean:

  command 'mist' from snap mist
  command 'hist' from deb loki
  command 'gist' from deb yorick
  command 'lift' from deb lift
  command 'last' from deb util-linux
  command 'flist' from deb mmh
  command 'flist' from deb nmh
  command 'dist' from deb mmh
  command 'dist' from deb nmh
  command 'mlist' from deb mblaze
  command 'klist' from deb heimdal-clients
  command 'klist' from deb krb5-user
  command 'bist' from deb bist

See 'snap info <snapname>' for additional versions.

root@ken-GE70-2PE:~# cd ./msi-keyboard-gui-linux-x64/resources/app/node_modules/msi-keyboard
-bash: cd: ./msi-keyboard-gui-linux-x64/resources/app/node_modules/msi-keyboard: No such file or directory
root@ken-GE70-2PE:~# 

The rest:
ken@ken-GE70-2PE:~/msi-keyboard-gui-linux-x64/resources/app/node_modules/msi-keyboard$ sudo npm install
[sudo] password for ken: 

> node-hid@0.7.2 install /home/ken/msi-keyboard-gui-linux-x64/resources/app/node_modules/msi-keyboard/node_modules/node-hid
> prebuild-install || node-gyp rebuild

prebuild-install WARN install EACCES: permission denied, mkdir '/home/ken/msi-keyboard-gui-linux-x64/resources/app/node_modules/msi-keyboard/node_modules/node-hid/build'
gyp ERR! configure error 
gyp ERR! stack Error: EACCES: permission denied, mkdir '/home/ken/msi-keyboard-gui-linux-x64/resources/app/node_modules/msi-keyboard/node_modules/node-hid/build'
gyp ERR! System Linux 4.15.0-23-generic
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/ken/msi-keyboard-gui-linux-x64/resources/app/node_modules/msi-keyboard/node_modules/node-hid
gyp ERR! node -v v8.11.3
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-hid@0.7.2 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-hid@0.7.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/ken/.npm/_logs/2018-06-23T14_46_15_282Z-debug.log
ken@ken-GE70-2PE:~/msi-keyboard-gui-linux-x64/resources/app/node_modules/msi-keyboard$ sudo node examples/color.js
/home/ken/msi-keyboard-gui-linux-x64/resources/app/node_modules/bindings/bindings.js:88
        throw e
        ^

Error: The module '/home/ken/msi-keyboard-gui-linux-x64/resources/app/node_modules/node-hid/build/Release/HID.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 54. This version of Node.js requires
NODE_MODULE_VERSION 57. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at Object.Module._extensions..node (module.js:681:18)
    at Module.load (module.js:565:32)
    at tryModuleLoad (module.js:505:12)
    at Function.Module._load (module.js:497:3)
    at Module.require (module.js:596:17)
    at require (internal/module.js:11:18)
    at bindings (/home/ken/msi-keyboard-gui-linux-x64/resources/app/node_modules/bindings/bindings.js:81:44)
    at Object.<anonymous> (/home/ken/msi-keyboard-gui-linux-x64/resources/app/node_modules/node-hid/nodehid.js:5:34)
    at Module._compile (module.js:652:30)
    at Object.Module._extensions..js (module.js:663:10)
ken@ken-GE70-2PE:~/msi-keyboard-gui-linux-x64/resources/app/node_modules/msi-keyboard$

stevelacy, 2018-06-23:
You missed the second command: ` cd ./msi-keyboard-gui-linux-x64` I posted: `cd msi-keyboard`
And the third command: `sudo npm install` I posted `npm install`
Delete the `msi-keyboard` folder and try again

f4exb, 2018-07-07:
I have the same issue with a recent GE73 series laptop. The vendor id and product id for the keyboard is one of these (don't know which one):

```lsusb | grep -i steelseries
Bus 001 Device 005: ID 1038:1122 SteelSeries ApS 
Bus 001 Device 003: ID 1038:1124 SteelSeries ApS 
```

stevelacy, 2018-07-09:
@f4exb does swapping out the product id [here](https://github.com/stevelacy/msi-keyboard/blob/master/lib/findKeyboard.js#L8) make any changes to the output?

f4exb, 2018-07-11:
Changing vendor and product IDs to the ones I have in the system give no error but no change on the keyboard lights either

stevelacy, 2018-08-22:
Moving to #53 as it is the same issue

closed 2018-08-22 by stevelacy