stephenlacy/msi-keyboard-gui #52: re-compiling or re-installing the module

unimilk456, 2018-08-07:
### Issue

**Describe your issue, comment or suggestion**
Hello
At the last step when I run command: 
`sudo npm start`
I've got error: 

> The module '/home/mike/Git/msi-keyboard-gui/dist/app/node_modules/node-hid/build/Release/HID.node'
> was compiled against a different Node.js version using
> NODE_MODULE_VERSION 54. This version of Node.js requires
> NODE_MODULE_VERSION 57. Please try re-compiling or re-installing
> the module (for instance, using `npm rebuild` or `npm install`).

Could you please explain me where I was wrong and what I had to do now ?
Thank you very much!

### System versions:
Ubuntu 18.04 

I run it on notebook msi ge73 rgb 8rf model  [https://us.msi.com/Laptop/GE73-Raider-RGB-Intel-8th-Gen.html](url)
```
nodejs:v8.11.3
npm: 6.3.0
libusb: libusb-1.0-0  libusb-0.1-4
```

stevelacy, 2018-08-08:
Please use the latest binaries https://github.com/stevelacy/msi-keyboard-gui/releases/tag/0.2.1
The issue is a version mismatch between node and election, you can also use node v8* and it should work

unimilk456, 2018-08-08:
> msi-keyboard-gui/releases/tag/0.2.1

Hello Steve!

Thank you for your fast answer!
I tried to run latest realize and got next error:

cannot open device with vendor id 0x1770 and product id 0xff00

So I run command:

> sudo msiklm list

and got next output:

> Device: SteelSeries KLC
> Device Vendor ID: 4152
> Device Product ID: 4386
> Device Serial Number: (null)
> Device Manufacturer: SteelSeries
> Device Path: 0001:0002:00
> Device Interface Number: 0
> Device Release Number: 553
> Device: SteelSeries KLC
> Device Vendor ID: 4152
> Device Product ID: 4386
> Device Serial Number: (null)
> Device Manufacturer: SteelSeries
> Device Path: 0001:0002:01
> Device Interface Number: 1
> Device Release Number: 553
>

So it looks like my keyboard doesn't support, does it?
Could you please to make necessary changes for supporting my keyboard?

Thank you very much!

Best wishes, Mike/

stevelacy, 2018-08-08:
I do not have a laptop with the new keyboard, so I can not test it.
What we can do is you could try to make the needed changes on your system, and figure out if the keyboard will work with the controller.

unimilk456, 2018-08-08:
Yes please, let's do it! :)  If you like I could provide you remote access
to my notebook.
I'm located in Moscow ()Russian Federation) so we could do it from 10-00 am
to  00-00 am local time.
my skype is unimilk456. Please  let me know  when you'll be ready :)

Be happy
Mikhail. :)

stevelacy, 2018-08-08:
I don't use skype, but I do use appear.in I should have free time next week, ping me by Wednesday if I don't contact you.

unimilk456, 2018-08-08:
okay, I will! What time for ping will good for you?

unimilk456, 2018-08-15:
Hello Steve!
I’m sorry I will not can be available today. Could  we please appoint other
day for checking keyboard?

Best wishes
Mikhail

ср, 8 авг. 2018 г. в 23:50, Mikhail <unimilk456@gmail.com>:

stevelacy, 2018-08-15:
Yes, early next week should work for me, email me to ensure I get the notification

stevelacy, 2018-08-22:
Moving to #53 as it uses the same keyboard

closed 2018-08-22 by stevelacy