stephenlacy/msi-keyboard-gui #53: Add support for msi gs65

thgn, 2018-08-21:
### Issue

Add support for msi gs65.

Executing:
```

$ sudo ./msi-keyboard-gui gives the output:

Error

cannot open device with vendor id 0x1770 and product id 0xff00
```

Did you try using sudo?
```

$ lsusb
Bus 001 Device 002: ID 1038:1122 SteelSeries ApS 
```

### System versions:

```
nodejs: nodejs-10.9.0-1
npm: 6.4.0
libusb: libusb-1.0.22-1
```

stevelacy, 2018-08-21:
Can you comment the output from this project? https://github.com/stevelacy/msi-keyboard-hid-test

That will show what node-hid is returning.

thgn, 2018-08-21:
Sure

```

> dev 1038 / 1122 : usage_page:5a10, usage:6766
> dev 1038 / 1122 : usage_page:1, usage:0
> [ { vendorId: 4152,
>     productId: 4386,
>     path: '/dev/hidraw0',
>     serialNumber: '',
>     manufacturer: 'SteelSeries',
>     product: 'SteelSeries KLC',
>     release: 553,
>     interface: 0,
>     usagePage: 23056,
>     usage: 26470 },
>   { vendorId: 4152,
>     productId: 4386,
>     path: '/dev/hidraw1',
>     serialNumber: '',
>     manufacturer: 'SteelSeries',
>     product: 'SteelSeries KLC',
>     release: 553,
>     interface: 1,
>     usagePage: 1 } ]
```

stevelacy, 2018-08-21:
What happens when you swap the vendorId and productId here? https://github.com/stevelacy/msi-keyboard/blob/master/lib/findKeyboard.js#L8

stevelacy, 2018-08-21:
What is your node version?

stevelacy, 2018-08-21:
Oh, this is why: `/usr/lib/node_modules/msi-keyboard-cli` the msi-keyboard-cli is not my project.

You don't install it, you clone it:
```
git clone https://github.com/stevelacy/msi-keyboard
cd msi-keyboard
npm install
```
Then edit the files

thgn, 2018-08-22:
Hi, I also tried yesterday git cloning and editing files with the same result :/. Is maybe  missing a dependency? I'm currently at work but I'll  post output at home.

stevelacy, 2018-08-22:
Did npm install complete without errors?

thgn, 2018-08-22:
I would say so...
```

sudo npm install
added 23 packages from 439 contributors and audited 2246 packages in 2.401s
found 0 vulnerabilities

```

stevelacy, 2018-08-22:
try it without sudo, so first remove them with `sudo rm -rf node_modules` and then `npm install`

thgn, 2018-08-22:
OK, I somehow  had screwed up npm. I deleted all configs (also I had a wrong  owner of node_modules in my home dir), removed msi-keyboard npm installed project and then reinstalled everything (included git cloning again).

Apparently I no longer have those errors, sorry for wasting your time I'm not very use to npm.

If I understood correctly from previous message you wanted to test the file lib/findkeyboard.js with the line:
`	var board = new hid.HID(4152, 4386);`
Now I get these error output.

**Before** changing:
```
sudo node blink.js 
/home/javier/node_modules/node-hid/nodehid.js:54
    this._raw = new (Function.prototype.bind.apply(binding.HID,
                ^

Error: cannot open device with vendor id 0x1770 and product id 0xff00
    at new HID (/home/javier/node_modules/node-hid/nodehid.js:54:17)
    at module.exports (/home/javier/msi-keyboard/lib/findKeyboard.js:8:14)
    at Object.<anonymous> (/home/javier/msi-keyboard/examples/blink.js:1:92)
    at Module._compile (internal/modules/cjs/loader.js:689:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:700:10)
    at Module.load (internal/modules/cjs/loader.js:599:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:538:12)
    at Function.Module._load (internal/modules/cjs/loader.js:530:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:742:12)
    at startup (internal/bootstrap/node.js:266:19)
```

**After** modifying I don't get any visible output in console or effect in the keyboard, I don't get the cursor back either.

stevelacy, 2018-08-22:
Sounds like the new keyboard is using a different hid interface

stevelacy, 2018-09-19:
This document goes over collecting and reverse engineering with usbpcap: https://github.com/charlysan/pyrfidhid/wiki/Reverse-Engineering-A-USB-HID-RFID-Reader-Writer
I looked at the packets capture and it seems that usbpcap might have split the requests into two instead of just the feature report with data.

dfredell, 2018-09-23:
I just found https://github.com/Askannz/msi-perkeyrgb#branch=master via https://wiki.archlinux.org/index.php/MSI_GS65#Lights and it worked as is. So, I will just use that.

stevelacy, 2018-09-24:
You could port this to node since you have access to the keyboard: https://github.com/Askannz/msi-perkeyrgb/blob/master/msi_perkeyrgb/protocol.py

neumartin, 2019-10-23:
Any update for GS65?

stevelacy, 2019-10-23:
No — I do not have a GS65 or access to one, thus I will close this issue

closed 2019-10-23 by stevelacy