stephenlacy/msi-keyboard-gui #55: Node-hid cannot seem to find my keyboard anymore: `cannot open device with vendor id 0x1770 and product id 0xff00`

bvobart, 2018-09-09:
### Issue

Hi, how has it been? I used to be able to use msi-keyboard and the GUI without any problems, but I recently reinstalled Ubuntu (same version, 18.04), and now I notice that msi-keyboard, or actually the underlying `node-hid`, can no longer seem to recognise my keyboard. I'm not sure if this might be a missing driver problem, but I was wondering whether you could help me shed light on this situation.

#### Symptoms:

Upon opening the GUI, I receive the error: `cannot open device with vendor id 0x1770 and product id 0xff00`

### System versions:

NodeJS: `8.11.4`
npm: `5.6.0`
libusb-dev: `2:0.1.12-31`
libusb-1.0-0-dev: `2:1.0.21-2`

### Troubleshooting steps

As a way of both trying to fix my problem, as well as helping to set up a set of standard troubleshooting questions and answers, which can then be placed in the ReadMe, I looked around to summarise the steps that can be taken to fix such a problem.

#### Did you use `sudo`?
Yes, I used `sudo` like there was no tomorrow.

#### Do you have the required versions `libusb-dev` and `libusb-1.0-0-dev` installed?

Yes, `dpkg -l | grep libusb` reports that versions `2:0.1.12-31` of `libusb-dev` and `2:1.0.21-2` of `libusb-1.0-0-dev` are installed. I presume these versions are high enough?

#### What is the output of [`msi-keyboard-hid-test`](https://github.com/stevelacy/msi-keyboard-hid-test)?:

Apparently it recognises my mouse, but no MSI or SteelSeries devices.

```
 ~/leds/msi-keyboard-hid-test > master > sudo node index.js
dev 46d / c332 : usage_page:10f8, usage:17e
dev 46d / c332 : usage_page:2d31, usage:2e31
[ { vendorId: 1133,
    productId: 49970,
    path: '/dev/hidraw0',
    serialNumber: '0E6336703330',
    manufacturer: 'Logitech',
    product: 'Gaming Mouse G502',
    release: 770,
    interface: 0,
    usagePage: 4344,
    usage: 382 },
  { vendorId: 1133,
    productId: 49970,
    path: '/dev/hidraw1',
    serialNumber: '0E6336703330',
    manufacturer: 'Logitech',
    product: 'Gaming Mouse G502',
    release: 770,
    interface: 1,
    usagePage: 11569,
    usage: 11825 } ]
```

#### What is the output of `lsusb`?

Interestingly, here we do see the desired 1770:ff00 device listed, though it does not have any name. That is consistent with the official `usb.ids` database however: https://usb-ids.gowdy.us/read/UD/1770

```
 ~/leds/msi-keyboard-hid-test > master > lsusb
Bus 002 Device 006: ID 1770:ff00  
Bus 002 Device 002: ID 8087:8000 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 8087:07dc Intel Corp. 
Bus 001 Device 005: ID 046d:c332 Logitech, Inc. 
Bus 001 Device 002: ID 8087:8008 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

#### Did you try purging and then reinstalling the package `xserver-xorg-input-all`?

Yes, didn't work.

stevelacy, 2018-09-10:
Thanks for debugging. I will fix the node-hid dependency and pin it on msi-keyboard

stevelacy, 2018-09-10:
I am on macos right now and am not able to build the release for windows/linux -- but I will build it shortly.

stevelacy, 2018-09-10:
Opening this until I push the release

stevelacy, 2018-09-10:
https://github.com/stevelacy/msi-keyboard-gui/releases/tag/v0.2.2
Published, let me know if it works for you, tested on debian.

closed 2018-09-10 by stevelacy

SaidMarar, 2019-01-29:
@stevelacy same problem with version 0.2.3 i remove it and install 0.2.2 and it works ! 0.2.3 does not support these changes

ansidev, 2020-04-01:
@stevelacy Currently, you must set driver type for node-hid or it won't find the MSI Steel Series Keyboard.

```
HID.setDriverType('libusb');
```

Or you can find a way to find it with hidraw default driver.

ansidev, 2020-04-01:
FYI
https://github.com/stevelacy/msi-keyboard/blob/master/lib/findKeyboard.js

stevelacy, 2020-04-01:
@ansidev PR welcome!

PR #56 (2018-09-10): Updates some dependencies: msi-keyboard to 0.4.1, electron to 1.8.8
Updating the `msi-keyboard` version fixes #55 and updating the Electron version to 1.8.8 fixed the node module version mismatch problem for me: 
```
The module '/home/bart/leds/fix/msi-keyboard-gui/dist/app/node_modules/node-hid/build/Release/HID.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 54. This version of Node.js requires
NODE_MODULE_VERSION 57. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
```

P.S. I'm using Node 8.11.4.