stephenlacy/msi-keyboard-gui #57: Not working on GT75

VBEnrico87, 2018-09-09:
### Issue

Hello,

I've tried it in the GT75 Titan 8RF

The keyboard seems to be found, but no effects on the keyboard.

What can I do to debug/help you to make it works?

Regards,

stevelacy, 2018-09-10:
I am assuming that all the late 2017 and 2018 versions are using a new keyboard with a different hid adapter (if any), none of the new laptops seem to work with this controller.
Right now I am tracking all the new keyboards and debugging in #53 
It will take a full reverse engineering of the new keyboard to make a new msi-keyboard module. I don't have access to a keyboard to attempt it.

VBEnrico87, 2018-09-10:
Hi,

I could just leave my PC on Windows with AnyDesk activated to let you reverse engineering the keyboard. What do you think?

Or if you want you could point me how to reverse engineering the keyboard (I'm a programmer)

Regards,

stevelacy, 2018-09-10:
This is where to start: https://steelseries.com/engine/msi 
On windows decipher what the steelseries engine uses to communicate with the keyboard, then replace the "real" hid device with a virtual device to capture the requests sent from the engine to the keyboard: 
- https://docs.microsoft.com/en-us/windows-hardware/drivers/hid/virtual-hid-framework--vhf-
- https://github.com/djpnewton/vmulti

Alternatively you could use node-hid and send/receive feature reports and data until something happens: https://github.com/node-hid/node-hid#reading-from-a-device

stevelacy, 2018-12-30:
Potentially you can convert these values to node: https://github.com/Askannz/msi-perkeyrgb/blob/master/msi_perkeyrgb/msi_keyboard.py#L7