stephenlacy/msi-keyboard-gui #62: Failed to start and install

iTechMouse, 2018-11-19:
### Issue
**During installation with npm this warning happens:
npm WARN optional Skipping failed optional dependency /chokidar/fsevents:
npm WARN notsup Not compatible with your operating system or architecture: fsevents@1.2.4

**Ignoring this, I did npm start and it failed outright:
npm ERR! Linux 4.15.0-39-generic
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "run" "build:app:linux"
npm ERR! node v8.10.0
npm ERR! npm  v3.5.2
npm ERR! code ELIFECYCLE
npm ERR! msi-keyboard-gui@0.2.2 build:app:linux: `npm rebuild --runtime=electron --target=1.8.8 --disturl=https://atom.io/download/atom-shell --npm_config_build_from_source=true && babel src/ --out-dir ./dist/app/src/ && cp -t ./dist/app/src/ ./src/index.html ./src/index.css && cp -rt ./dist/app/ index.js package.json node_modules LICENSE`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the msi-keyboard-gui@0.2.2 build:app:linux script 'npm rebuild --runtime=electron --target=1.8.8 --disturl=https://atom.io/download/atom-shell --npm_config_build_from_source=true && babel src/ --out-dir ./dist/app/src/ && cp -t ./dist/app/src/ ./src/index.html ./src/index.css && cp -rt ./dist/app/ index.js package.json node_modules LICENSE'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the msi-keyboard-gui package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     npm rebuild --runtime=electron --target=1.8.8 --disturl=https://atom.io/download/atom-shell --npm_config_build_from_source=true && babel src/ --out-dir ./dist/app/src/ && cp -t ./dist/app/src/ ./src/index.html ./src/index.css && cp -rt ./dist/app/ index.js package.json node_modules LICENSE
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs msi-keyboard-gui
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls msi-keyboard-gui
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/sam/msi-keyboard-gui/npm-debug.log

npm ERR! Linux 4.15.0-39-generic
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "start"
npm ERR! node v8.10.0
npm ERR! npm  v3.5.2
npm ERR! code ELIFECYCLE
npm ERR! msi-keyboard-gui@0.2.2 start: `npm run build:app:linux && electron ./dist/app`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the msi-keyboard-gui@0.2.2 start script 'npm run build:app:linux && electron ./dist/app'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the msi-keyboard-gui package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     npm run build:app:linux && electron ./dist/app
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs msi-keyboard-gui
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls msi-keyboard-gui
npm ERR! There is likely additional logging output above.

### System versions:
Ubuntu 18.04.1 on MSI Stealth Pro GS63VR 6RF
```
nodejs: v8.10
npm: v3.5.2
libusb:
```

Wispeon, 2018-11-19:
***Note*** 
I'm not the most proficient with Linux, so this is probably my own user error.

stevelacy, 2018-11-19:
_Ignore all warnings from npm on almost any npm project_

Those errors do not actually say what the issue is, just install from the latest release: https://github.com/stevelacy/msi-keyboard-gui/releases

stevelacy, 2018-11-19:
If you can attach the error log that might help.

Wispeon, 2018-11-20:
I tried doing the install from latest release, and after using tar -xf (successfully) I used the sudo ./msi-keyboard-gui-linux-x64/msi-keyboard-gui command.

 sudo ./msi-keyboard-gui-linux-x64/msi-keyboard-gui
[sudo] password for sam: 
./msi-keyboard-gui-linux-x64/msi-keyboard-gui: error while loading shared libraries: libgconf-2.so.4: cannot open shared object file: No such file or directory

Again, could be my fault since I'm slowly learning Linux.

The attachment is the debug log from npm.
[npm-debug.log](https://github.com/stevelacy/msi-keyboard-gui/files/2597963/npm-debug.log)

stevelacy, 2018-11-20:
That error tells you what you need to install
https://stackoverflow.com/questions/37624225/shared-libraries-libgconf-2-so-4-is-missing

7tails, 2018-11-20:
Fixed it, problem solved. Thank you apologizes for me being a beginner.

closed 2018-11-20 by 7tails

stevelacy, 2018-11-20:
Glad to help