stephenlacy/msi-keyboard-gui #65: libgtk-x11 error Ubuntu 19.04

Soxcks, 2019-07-11:
### Issue

On a fresh install of Ubuntu 19.04. Followed the Linux instructions exactly. 
1. Download the latest release (msi-keyboard-gui-linux-0.2.2.tar.gz)
2. run 
```
tar -xf ./msi-keyboard-gui-linux-0.2.2.tar.gz
sudo ./msi-keyboard-gui-linux-x64/msi-keyboard-gui
```
3. Encounter error:
`./msi-keyboard-gui: error while loading shared libraries: libgtk-x11-2.0.so.0: cannot open shared object file: No such file or directory`

### System versions:

```
nodejs: v12.6.0
npm: 6.9.0
libusb: 
dpkg --list | grep libusb
ii  libgusb2:amd64                             0.3.0-1                             amd64        GLib wrapper around libusb1
ii  libusb-1.0-0:amd64                         2:1.0.22-2                          amd64        userspace USB programming library
ii  libusbmuxd4:amd64                          1.1.0~git20181007.07a493a-1         amd64        USB multiplexor daemon for iPhone and iPod Touch devices - library
```

stevelacy, 2019-07-11:
check if this works: https://askubuntu.com/questions/1085282/missing-libgtk-x11-2-0-so-0-in-18-04?noredirect=1&lq=1

Soxcks, 2019-07-12:
After running `sudo apt install --reinstall libgtk2.0-0`, I now get a new error: `./msi-keyboard-gui: error while loading shared libraries: libgconf-2.so.4: cannot open shared object file: No such file or directory`

Soxcks, 2019-07-12:
Sorry I jumped the gun. I have fixed this and have the app running. 

To fix the libgtk error:
`sudo apt install --reinstall libgtk2.0-0`

To fix the libgconf error:
`sudo apt install libgconf-2-4`

Thanks for your help!

stevelacy, 2019-07-12:
nice!

closed 2019-07-12 by stevelacy