stephenlacy/msi-keyboard-gui #71: Install failed

Gifhshady, 2021-01-08:
### Issue

**Failed installation following instructions**

### System versions: Mint 20

```
nodejs: 10
npm:
libusb:
```
Just followed instructions:

Install
Node.js 6+ is REQUIRED (got it from Synaptic)
From prebuilt binaries:

Download the latest from the releases (got your last release msi-keyboard-gui-0.2.3.tar.gz)

Linux:

$ tar -xf ./msi-keyboard-gui-linux*.tar.gz (doesnt work)
$ sudo ./msi-keyboard-gui-linux-x64/msi-keyboard-gui  (doesnt work)

I'm not very familiar with this kind of installation so I'm begging you, please update installation instructions or release  a .deb optional installation file.
Thank you for your nice work!
Gifh

stevelacy, 2021-01-08:
If `tar` does not work then definitely everything else won't work.
Please explain why it "doesn't work" or I can't help you.

Gifhshady, 2021-01-09:
Thanks for reply SteveLacy. 
`tar` failed to work due to different name of the file downloaded, but even if I manage to extract all files, the second line give something like:
`Bash: command not found` and I really dont know what to do.

stevelacy, 2021-01-09:
List _Every single command_ you are typing with screenshots if possible.

The commands to run are very simple:

Download: https://github.com/stevelacy/msi-keyboard-gui/releases/download/v0.2.2/msi-keyboard-gui-linux-0.2.2.tar.gz
```
tar -xf msi-keyboard-gui-0.2.2.tar.gz
sudo ./msi-keyboard-gui-linux-x64/msi-keyboard-gui

```

Gifhshady, 2021-01-09:
Apart different file name in your first tar command, that I can solve, the second line gives the same error.
[image]
Also, just a curiosity, why not version 0.2.3?

Gifhshady, 2021-01-09:
[image]
and this is the result if I change in the right directory

Gifhshady, 2021-01-09:
Ok, after analyzing first screenshot, I noted the missing shared library, so I did
`sudo apt install libgconf2-4`, repeat your last sudo and all work greatly, thanks a lot!
Very nice utility!

stevelacy, 2021-01-10:
Awesome, glad it worked!

closed 2021-01-10 by stevelacy