stephenlacy/msi-keyboard-gui #76: (11/2022) FYI - libgenconf-2-4 will be needed for install

AtomicRobotMan0101, 2022-11-12:
**FYI for future peoples....**

Install **Node.js** and **libgenconf-2-4** as the install might bomb out.

The LibGen is not present on Linux Mint 21 and should be installed via your terminal using:

`sudo apt install libgconf-2-4`

Thought to advise as many gamers using Linux would not be aware of how to solve this issue.

Otherwise, can confirm that the project still works perfectly. In fact, I'm super amazed at how well it does. Kudos.

[image]

stevelacy, 2022-11-12:
Good to know, thanks. Want to send a PR to update the install instructions?

kwolf-22, 2025-03-30:
Just installed this on an MSI GS73VR running a clean install of Ubuntu 24.04.2 LTS.  Got it working with these two install steps:

**Install Node.js:**
Snap is built into Ubuntu 24.04.2 
`sudo snap install node --classic`

**Install libgconf-2-4:**
Download the following packages:
```
wget http://ftp.it.debian.org/debian/pool/main/g/gconf/gconf2_3.2.6-8_amd64.deb
wget http://ftp.it.debian.org/debian/pool/main/g/gconf/libgconf-2-4_3.2.6-8_amd64.deb
wget http://ftp.it.debian.org/debian/pool/main/g/gconf/gconf2-common_3.2.6-8_all.deb
wget http://ftp.it.debian.org/debian/pool/main/g/gconf/gconf-service_3.2.6-8_amd64.deb
wget http://ftp.it.debian.org/debian/pool/main/o/openldap/libldap-2.5-0_2.5.13%2bdfsg-5_amd64.deb
```
Then install the packages in the following order:
```
sudo dpkg -i gconf2-common_3.2.6-8_all.deb 		// Install gconf2-common first
sudo dpkg -i libgconf-2-4_3.2.6-8_amd64.deb 		// Then libgconf
sudo dpkg -i libldap-2.5-0_2.5.13+dfsg-5_amd64.deb 	// Then libldap
sudo dpkg -i gconf-service_3.2.6-8_amd64.deb 		// Then the gconf service
sudo dpkg -i gconf2_3.2.6-8_amd64.deb 		 	// Finally gconf.
```
The order is important.

After that, everything works great.  Thanks for developing this software!  Let me know if you want the instructions updated with these details.  I'm happy to contribute.

stephenlacy, 2025-04-05:
I'm glad this project is still getting used!
Do you happen to have the install error? Running `apt --fix-broken install` after `sudo apt install libgconf-2-4` should be all that is required, downloading from the pool should not be needed.